MicrochipTech/cryptoauthlib #205: Error in pkcs11_init:1439: PKCS #11 error

pavithran93, 2021-01-23:
I followed the steps in 
https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup. 

When I run p11tool --list-all I am receiving below response
[image]

When I run p11tool --list-all --provider /usr/lib/libcryptoauth.so I getting this response
[image]

What's the difference between these two commands. Why my outcome is different?

Next I am trying to initialize the chip using the command p11tool --initialize "pkcs11:serial=10A592C2E036" --label appzgate. I am getting the error 
Error in pkcs11_init:1439: PKCS #11 error.
[image]

As per the tutorial I am following it supposed to initialize the chip. But I am getting the error, Please assist me to resolve this to proceed further

bryan-hunt, 2021-01-27:
So the issue I'm seeing from the above are as follows

1) use the latest release that has been published - in your case to minimize changes use tag 20210121 which is the most recent python version - this will help if you need to run python scripts to verify anything

2) You are seeing the two tokens because your p11-kit set up is allowing for it. You can modify your p11-kit configuration from merge to only which instead of merging the module configuration from root with the module configuration in your user directory it will instead only use the user configuration.

3) You are seeing a private and public key from the device - this is an indication that the device is configured and locked already. Trying to initialize an already initialized device will fail.

pavithran93, 2021-01-27:
Hi bryan,
             Thanks for your reply
             After I restart the device. I can initialize the token
[image]
            
              Once I initialize the totally showing three objects. Weather can I remove the key after initialization?
              
              Weather I can generate the LUKS key file in the device ATECC608A. Can access that key at initramfs window?

pavithran93, 2021-01-27:
[image]

You check the above picture. 
1) I am generating key file in the pen drive
2) Adding that key file from pen drive to my LUKS encrypted patrician
3) Referring the pen drive UUID  to cryptab for auto unlock the SD card

I am trying to do similar things with ATECC608A. Because storing the key file in pen drive is not safer

pavithran93, 2021-05-27:
Hi Bryan,
               I successfully initialized and verified the certificate.
               I need to know how to add keys to ATECC608A.
               How to get back stored keys.
               How to lock the ATECC608A corresponding to hardware (Raspberry Pi). By this we will ensure the ATECC608A should work only on specific hardware device.
               Please assist me on this.

pavithran93, 2021-06-01:
Thank you.
                 Issue got solved

closed 2021-06-01 by pavithran93