MicrochipTech/cryptoauthlib #211: Cryptoauthlib, STM32, arm-none-eabi-gcc, libopenCM3 and makefiles.

krish2487, 2021-03-17:
Hello folks,

I am trying to use cryptoauthlib with an STM32 controller and libopenCM3 HAL.
I have run into a roadblock while trying to integrate the library into my application.
I have started with the libopenCM3 template from [here](https://github.com/libopencm3/libopencm3-template)
and I have renamed the "my-common-code" to "shared_libs" and "my-project" to "cryptoauth_test". I have then cloned the cryptoauthlib into the "shared_libs" folder and created the "build" folder under the "cryptoauthlib" folder.

So my folder structure looks like this.
`.

├── cryptoauth_test
│   └── bin
├── libopencm3
.....
│   ├── include
│   │   ├── libopencm3
│   │   │   ├── cm3
│   │   │   ├── dispatch
│   │   │   ├── efm32
│   │   │   ├── ethernet
│   │   │   ├── lm3s
│   │   │   ├── lm4f
│   │   │   ├── lpc13xx
│   │   │   ├── lpc17xx
│   │   │   ├── lpc43xx
│   │   │   ├── msp432
│   │   │   ├── sam
│   │   │   ├── stm32
│   │   │   ├── usb
│   │   │   └── vf6xx
│   │   └── libopencmsis
│   │       ├── dispatch
│   │       ├── efm32
│   │       ├── lm3s
│   │       ├── lpc13xx
│   │       ├── lpc17xx
│   │       ├── lpc43xx
│   │       ├── msp432
│   │       ├── sam
│   │       ├── stm32
│   │       └── vf6xx
└── shared_libs
    ├── cryptoauthlib
    │   ├── app
    │   │   ├── api_206a
    │   │   ├── ip_protection
    │   │   ├── pkcs11
    │   │   ├── secure_boot
    │   │   └── tng
    │   ├── build
    │   │   ├── CMakeFiles
    │   │   └── lib
    │   ├── harmony
    │   │   ├── config
    │   │   ├── src
    │   │   └── templates
    │   ├── lib
    │   │   ├── atcacert
    │   │   ├── calib
    │   │   ├── crypto
    │   │   ├── hal
    │   │   ├── host
    │   │   ├── jwt
    │   │   ├── mbedtls
    │   │   ├── openssl
    │   │   ├── pkcs11
    │   │   └── wolfssl
    │   ├── python
    │   │   ├── cryptoauthlib
    │   │   └── tests
[... 290 lines cut ...]
  210 | static void cdcacm_data_tx_cb(usbd_device *usbd_dev, uint8_t ep)
      |             ^~~~~~~~~~~~~~~~~
  CC    ../shared_libs/delay.c
arm-none-eabi-gcc -DATCA_HAL_STM32  -std=c11 -ggdb3 -mcpu=cortex-m3 -mthumb -msoft-float -fno-common  -ffunction-sections -fdata-sections -fstack-usage -Wextra -Wshadow -Wno-unused-variable -Wimplicit-function-declaration -Wredundant-decls -Wstrict-prototypes -Wmissing-prototypes  -MD -Wall -Wundef -I. -I../shared_libs -I../shared_libs/cryptoauthlib/lib -I. -I../libopencm3/include -I. -I../shared_libs -I../shared_libs/cryptoauthlib/lib -I. -I../libopencm3/include  -DSTM32F1 -DSTM32F103RCT6 -I../libopencm3/include -o bin/delay.o -c ../shared_libs/delay.c
  GENLNK  stm32f103rct6
arm-none-eabi-gcc -E -mcpu=cortex-m3 -mthumb -msoft-float -DSTM32F1 -DSTM32F103RCT6 -D_ROM=256K -D_RAM=48K -D_ROM_OFF=0x08000000 -D_RAM_OFF=0x20000000 -P -E ../libopencm3/ld/linker.ld.S > generated.stm32f103rct6.ld
  LD    crypto_test.elf
arm-none-eabi-gcc -L ../shared_libs/cryptoauthlib/build/lib/ -lcryptoauth -Tgenerated.stm32f103rct6.ld -L../libopencm3/lib -nostartfiles -mcpu=cortex-m3 -mthumb -msoft-float -specs=nano.specs  -Wl,--build-id=uuid -Wl,-Map=crypto_test.map -L../libopencm3/lib bin/main.o bin/delay.o -lopencm3_stm32f1 -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -o crypto_test.elf
/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: bin/main.o: in function `main':
/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/cryptoauth_test/main.c:55: undefined reference to `atcab_init'
/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/cryptoauth_test/main.c:59: undefined reference to `atcab_random'
/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/cryptoauth_test/main.c:68: undefined reference to `cfg_ateccx08a_i2c_default'
collect2: error: ld returned 1 exit status
make: *** [crypto_test.elf] Error 1`

It looks like the compiler is linking the cryptoauthlib but is yet not finding the actual source for the functions it is complaining about as seen from this line here.
` LD    crypto_test.elf
arm-none-eabi-gcc -L ../shared_libs/cryptoauthlib/build/lib/ -lcryptoauth -Tgenerated.stm32f103rct6.ld -L../libopencm3/lib -nostartfiles -mcpu=cortex-m3 -mthumb -msoft-float -specs=nano.specs  -Wl,--build-id=uuid -Wl,-Map=crypto_test.map -L../libopencm3/lib bin/main.o bin/delay.o -lopencm3_stm32f1 -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -o crypto_test.elf`

I am not sure how to resolve this issue. I would be extremely grateful if someone more knowledgeable can point out what I am doing wrong and how I can fix this and get the library to link and successfully compile.

Or, if I am going about it the wrong way and there is in fact an easier way to integrate the library into a makefile project.. I am game for it as well. 

Thanks in advance folks. 
I apologize for the long post, but I feel like the more information I give about what and how I have done my work, will eable others to help me better. :-)

bryan-hunt, 2021-03-17:
Did you set up the environment to enable cross compilation for CMake with the correct paths so it picks up correct compiler and library paths? As for the missing symbols you can see that they are are only present when the configuration allows for it: https://github.com/MicrochipTech/cryptoauthlib/blob/main/lib/atca_cfgs.c#L40-L58 - In other words in order to have the I2C default structure defined you need to enable ATCA_HAL_I2C

krish2487, 2021-03-17:
@bryan-hunt 
Thank you for the prompt response.
I have not setup the environment for cross compilation.
In the this page [here](https://github.com/MicrochipTech/cryptoauthlib/wiki/Configuration) in the first line it states that cmake configuration does not support cross compilation.

If I can trouble you, how would I go about doing the cross compilation 
As I understand it, it should be as simple as 

` CC=arm-none-eabi-gcc CXX=arm-none-eabi-g++ cmake  -DATCA_BUILD_SHARED_LIBS=OFF -DATCA_HAL_I2C=ON ../`

while building the cryptoauthlib isnt it??

Secondly, if you look at the commands I have given for building the cryptoauthlib, I did indeed define the "ATCA_HAL_I2C" parameter

```
cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_HAL_I2C=ON -DATCA_BUILD_SHARED_LIBS=OFF ../  
 cmake --build .    

```

but looking at the link you provided I did not define the "ATCA_ECC_SUPPORT" parameter. I can rebuilding the library with those specific define flags and try again.. but I suspect it might still not work, if it boils down to a cross compilation issue

bryan-hunt, 2021-03-17:
yeah I believe that generally the build files you're getting are jumbled. I would suggest setting up your own makefiles as one project. You can copy the atca_config.h file you created already as a start but you'll have to make some adjustments. Cryptoauthlib needs timing functions as well - the linux timing functions that would be included using the above process won't work in your STM32.

krish2487, 2021-03-17:
If i can trouble you for some specific help, can you give a sample project that I can work off from @bryan-hunt .

I see from this post  that you have posted a rather nice example for compiling and integrating the cryptoauthlib for ESP32.
https://github.com/MicrochipTech/cryptoauthlib/issues/5#issuecomment-433541966

I was wondering, if you can suggest something similar. I understand that I would be subjecting you to some inconvenience and extra work, but it would mean a lot to me and I hope provide inspiration to many others to integrate the cryptoauthlib into their projects and platforms.

Thank you in advance again!! :-)

krish2487, 2021-03-18:
@bryan-hunt , I have tried to cross compile the cryptoauthlib using the arm-none-eabi-gcc compiler like so
`CC=arm-none-eabi-gcc cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_HAL_I2C=ON -DATCA_BUILD_SHARED_LIBS=OFF -DATCA_ECC_SUPPORT=ON ../`

and it looks like cmake is unable to cross compile with this compiler. I get this error

```
 CC=arm-none-eabi-gcc cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_HAL_I2C=ON -DATCA_BUILD_SHARED_LIBS=OFF -DATCA_ECC_SUPPORT=ON ../
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 9.3.1
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- Check for working C compiler: /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/local/Cellar/cmake/3.19.6/share/cmake/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make cmTC_c2c8c/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_c2c8c.dir/build.make CMakeFiles/cmTC_c2c8c.dir/build
    Building C object CMakeFiles/cmTC_c2c8c.dir/testCCompiler.c.o
    /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-gcc   -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk -o CMakeFiles/cmTC_c2c8c.dir/testCCompiler.c.o -c /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_c2c8c
    /usr/local/Cellar/cmake/3.19.6/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c2c8c.dir/link.txt --verbose=1
    /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-gcc  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_c2c8c.dir/testCCompiler.c.o -o cmTC_c2c8c
    /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol arch_paths_first; defaulting to 0000000000008018
    /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): in function `exit':
    exit.c:(.text.exit+0x2c): undefined reference to `_exit'
    collect2: error: ld returned 1 exit status
    make[1]: *** [cmTC_c2c8c] Error 1
    make: *** [cmTC_c2c8c/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/build/CMakeFiles/CMakeOutput.log".
See also "/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/build/CMakeFiles/CMakeError.log".
```

Is there anything else I can try on my end to compile the library?? 
Thanks in advance!! :-)

krish2487, 2021-03-18:
Ok.. Some progress and I think this might provide some hint as to why it is not working.. To fix the issue of cross compilation, I have done some reading and I modified the CMakeLists.txt to reflect the toolchain I am using for STM32. I have added this section at the beginning of the file just after the version setting.

```
set(CMAKE_SYSTEM_NAME      Generic)
set(CMAKE_SYSTEM_VERSION   1)
set(CMAKE_SYSTEM_PROCESSOR arm-eabi)

set(CMAKE_AR             /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-ar CACHE PATH "" FORCE)
set(CMAKE_ASM_COMPILER   /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-as CACHE PATH "" FORCE)
set(CMAKE_C_COMPILER     /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-gcc CACHE PATH "" FORCE)
set(CMAKE_CXX_COMPILER   /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-g++ CACHE PATH "" FORCE)
set(CMAKE_LINKER         /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-ld CACHE PATH "" FORCE)
set(CMAKE_OBJCOPY        /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-objcopy CACHE PATH "" FORCE)
set(CMAKE_RANLIB         /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-ranlib CACHE PATH "" FORCE)
set(CMAKE_SIZE           /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-size CACHE PATH "" FORCE)
set(CMAKE_STRIP          /Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/arm-none-eabi-strip CACHE PATH "" FORCE)

set(TOOCHAIN_LIB_DIR " ~/gcc-arm-none-eabi/lib/")

```

I have sorta taken inspiration from this file - [here](https://github.com/adrienbailly/STM32-CMake-CodeSourcery/blob/master/CMake/Toolchain-STM32-CodeSourcery.cmake)

and now the first step works okay and this is the output I see

```
 cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_HAL_I2C=ON -DATCA_BUILD_SHARED_LIBS=OFF -DATCA_ECC_SUPPORT=ON ../
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is AppleClang 12.0.0.12000026
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for malloc
-- Looking for malloc - found
-- Looking for free
-- Looking for free - found
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    ATCA_ECC_SUPPORT

-- Build files have been written to: /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/build
```
However when I run the 
`cmake --build .` it throws a bunch of errors. This is what I see on the console.

```
cmake --build .
[  1%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_basic.c.o
/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/lib/atca_basic.c: In function 'atcab_write_config_zone':
/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/lib/atca_basic.c:3916:63: warning: passing argument 2 of 'calib_ecc204_write_config_zone' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 3916 |             status = calib_ecc204_write_config_zone(_gDevice, config_data);
      |                                                               ^~~~~~~~~~~
In file included from /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/lib/cryptoauthlib.h:109,
                 from /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/lib/atca_basic.h:34,
                 from /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/lib/atca_basic.c:29:
/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/shared_libs/cryptoauthlib/lib/calib/calib_basic.h:206:72: note: expected 'uint8_t *' {aka 'unsigned char *'} but argument is of type 'const uint8_t *' {aka 'const unsigned char *'}
  206 | ATCA_STATUS calib_ecc204_write_config_zone(ATCADevice device, uint8_t* config_data);
      |                                                               ~~~~~~~~~^~~~~~~~~~~
[  2%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_cfgs.c.o
[... 342 lines cut ...]
/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/include/sys/mman.h:238:30: error: parameter name omitted
  238 | void *  mmap(void *, size_t, int, int, int, off_t) __DARWIN_ALIAS(mmap);
      |                              ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/include/sys/mman.h:238:35: error: parameter name omitted
  238 | void *  mmap(void *, size_t, int, int, int, off_t) __DARWIN_ALIAS(mmap);
      |                                   ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/include/sys/mman.h:238:40: error: parameter name omitted
  238 | void *  mmap(void *, size_t, int, int, int, off_t) __DARWIN_ALIAS(mmap);
      |                                        ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/include/sys/mman.h:238:45: error: parameter name omitted
[... cut ...]
make[2]: *** [lib/CMakeFiles/cryptoauth.dir/hal/hal_linux.c.o] Error 1
make[1]: *** [lib/CMakeFiles/cryptoauth.dir/all] Error 2
make: *** [all] Error 2
```

I think this is a step in right direction but I need some help fixing these issues. It looks like the compilation is failing at the hal_linux.c file and a bunch of other "conflicting types" errors.  

Is there anything I have done wrong on the CMakeLists.txt file that I should correct or exclude certain files from compiling? If so, how??

Thanks!! :-)

krish2487, 2021-03-18:
Ok, on a hunch I decided to compile natively again and test a hypothesis. I removed the cross compiling options in the CMakeLists.txt and reverted back to the stock version.

This is the command I ran
````
 cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_HAL_I2C=ON -DATCA_BUILD_SHARED_LIBS=OFF -DATCA_ECC_SUPPORT=ON -DATCA_HAL_STM32=ON ../
cmake --build .
````
and it built the library successfully.

I navigated to the lib folder under the build folder and ran
`ar -t libcryptoauth.a`
to look at the library contents and this is the console output

```
ar -t libcryptoauth.a
__.SYMDEF SORTED
atca_basic.c.o
atca_cfgs.c.o
atca_debug.c.o
atca_device.c.o
atca_helpers.c.o
atca_iface.c.o
atca_utils_sizes.c.o
calib_aes.c.o
calib_aes_gcm.c.o
calib_basic.c.o
calib_checkmac.c.o
calib_command.c.o
calib_counter.c.o
calib_derivekey.c.o
calib_ecdh.c.o
calib_execution.c.o
calib_gendig.c.o
calib_genkey.c.o
calib_hmac.c.o
calib_info.c.o
calib_kdf.c.o
calib_lock.c.o
calib_mac.c.o
calib_nonce.c.o
calib_privwrite.c.o
calib_random.c.o
calib_read.c.o
calib_secureboot.c.o
calib_selftest.c.o
calib_sha.c.o
calib_sign.c.o
calib_updateextra.c.o
calib_verify.c.o
calib_write.c.o
atca_host.c.o
api_206a.c.o
atcacert_client.c.o
atcacert_date.c.o
atcacert_def.c.o
atcacert_der.c.o
atcacert_host_hw.c.o
atcacert_host_sw.c.o
atcacert_pem.c.o
atca_crypto_hw_aes_cbc.c.o
atca_crypto_hw_aes_cbcmac.c.o
atca_crypto_hw_aes_ccm.c.o
atca_crypto_hw_aes_cmac.c.o
atca_crypto_hw_aes_ctr.c.o
atca_crypto_sw_ecdsa.c.o
atca_crypto_sw_rand.c.o
atca_crypto_sw_sha1.c.o
atca_crypto_sw_sha2.c.o
sha1_routines.c.o
sha2_routines.c.o
atca_jwt.c.o
tflxtls_cert_def_4_device.c.o
tng_atca.c.o
tng_atcacert_client.c.o
tng_root_cert.c.o
tnglora_cert_def_1_signer.c.o
tnglora_cert_def_2_device.c.o
tnglora_cert_def_4_device.c.o
tngtls_cert_def_1_signer.c.o
tngtls_cert_def_2_device.c.o
tngtls_cert_def_3_device.c.o
atca_hal.c.o
hal_linux.c.o
```

Two issues popped out to me.

1.  The library indeed contains the object files for atca_basic and atca_cfgs which contain the atcab_init and the cfg_ateccx08a_i2c_default definitions. 

```
make V=1
  CC    main.c
arm-none-eabi-gcc -DATCA_HAL_STM32  -std=c11 -ggdb3 -mcpu=cortex-m3 -mthumb -msoft-float -fno-common  -ffunction-sections -fdata-sections -fstack-usage -Wextra -Wshadow -Wno-unused-variable -Wimplicit-function-declaration -Wredundant-decls -Wstrict-prototypes -Wmissing-prototypes  -MD -Wall -Wundef -I. -I../shared_libs -I../shared_libs/cryptoauthlib/lib -I. -I../libopencm3/include -I. -I../shared_libs -I../shared_libs/cryptoauthlib/lib -I. -I../libopencm3/include  -DSTM32F1 -DSTM32F103RCT6 -I../libopencm3/include -o bin/main.o -c main.c
In file included from main.c:9:
usb.h:210:13: warning: 'cdcacm_data_tx_cb' defined but not used [-Wunused-function]
  210 | static void cdcacm_data_tx_cb(usbd_device *usbd_dev, uint8_t ep)
      |             ^~~~~~~~~~~~~~~~~
  CC    ../shared_libs/delay.c
arm-none-eabi-gcc -DATCA_HAL_STM32  -std=c11 -ggdb3 -mcpu=cortex-m3 -mthumb -msoft-float -fno-common  -ffunction-sections -fdata-sections -fstack-usage -Wextra -Wshadow -Wno-unused-variable -Wimplicit-function-declaration -Wredundant-decls -Wstrict-prototypes -Wmissing-prototypes  -MD -Wall -Wundef -I. -I../shared_libs -I../shared_libs/cryptoauthlib/lib -I. -I../libopencm3/include -I. -I../shared_libs -I../shared_libs/cryptoauthlib/lib -I. -I../libopencm3/include  -DSTM32F1 -DSTM32F103RCT6 -I../libopencm3/include -o bin/delay.o -c ../shared_libs/delay.c
  GENLNK  stm32f103rct6
arm-none-eabi-gcc -E -mcpu=cortex-m3 -mthumb -msoft-float -DSTM32F1 -DSTM32F103RCT6 -D_ROM=256K -D_RAM=48K -D_ROM_OFF=0x08000000 -D_RAM_OFF=0x20000000 -P -E ../libopencm3/ld/linker.ld.S > generated.stm32f103rct6.ld
  LD    crypto_test.elf
arm-none-eabi-gcc -L ../shared_libs/cryptoauthlib/build/lib/ -lcryptoauth -Tgenerated.stm32f103rct6.ld -L../libopencm3/lib -nostartfiles -mcpu=cortex-m3 -mthumb -msoft-float -specs=nano.specs  -Wl,--build-id=uuid -Wl,-Map=crypto_test.map -L../libopencm3/lib bin/main.o bin/delay.o -lopencm3_stm32f1 -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -o crypto_test.elf
/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: bin/main.o: in function `main':
/Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/cryptoauth_test/main.c:55: undefined reference to `atcab_init'
/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/cryptoauth_test/main.c:59: undefined reference to `atcab_random'
/Users/srikrishnachaitanyanarumanchi/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: /Users/srikrishnachaitanyanarumanchi/Workspace/libopencm3/test/cryptoauth_test/main.c:68: undefined reference to `cfg_ateccx08a_i2c_default'
collect2: error: ld returned 1 exit status
make: *** [crypto_test.elf] Error 1

```

So the library definitely has these definitions present.

2. The last line of the ar output seems to indicate that hal used is the hal_linux and not the stm32 custom hal, inspite of me explicitly passing the -DATCA_HAL_STM32 flag to cmake when building the library as seen above.

@bryan-hunt  Is there anything else I can try from my end to get it to link successfully?

bryan-hunt, 2021-03-18:
Going back to my previous comment - write your own makefiles to include the proper sources in your project. You also might try using a development IDE that will make them for you. There many open source projects out there for MCU development.

https://eclipse-embed-cdt.github.io/ https://platformio.org/  and stm32cubeide all come to mind as options

krish2487, 2021-03-18:
Ok, if I am understanding you correctly, I should not try and link the library separately but compile the sources for the cryptoauthlib as a part of my source files manually? Is that right? 

and where should I look for the appropriate source files? Can I simply include the source files for the list of object files i get when i ran the ar command?

bryan-hunt, 2021-03-18:
This is outside the scope of the kind of support we're able to provide short of suggesting an embedded development tutorial. We have several premade project to demonstrate the features of the device - you can find them in the MPLAB Harmony application projects and in the Trust Platform Development Suite.

krish2487, 2021-03-19:
@bryan-hunt  
Thank you, but I am already sold on the features of the product. I just need help using it in my application. :-)
Nevertheless, thank you for the help so far. I guess, I have to look for help elsewhere.

closed 2021-03-19 by krish2487

krish2487, 2021-03-19:
@bryan-hunt  Ok.. I followed your advice and started by adding all the necessary source files required to my existing makefile, which happened to be the files in the lib/, calib/, crypto/ and crypto/hashes/ folders and the binary compiles successfully now. :-) Now I have to work on separating the library build from the rest of the firmware. 
Thank you!!