MicrochipTech/cryptoauthlib #214: Function atcab_init returns error code 240: ATCA_COMM_FAIL with i2c communication

MrMelon1232, 2021-03-26:
Hi,

I am currently using this library with the ATECC608A chip along with an esp32. I have been trying to use the function atcab_init but it keeps returning error code 240 which is ```ATCA_COMM_FAIL```. I have been following this example here: https://github.com/kmwebnet/ECC608-AzureIoT-Connect. 
Here is my configuration: 
```
    ATCA_STATUS err;
    if(i2c_master_init() == ESP_OK)
        printf("i2c master init success!\n");
    ATCAIfaceCfg atecc608 ={
        .iface_type = ATCA_I2C_IFACE,
        .devtype = ATECC608A,
        .atcai2c.slave_address = ATECC608_ADDR,
        .atcai2c.bus = 1,
        .atcai2c.baud = 100000,
        .wake_delay = 1500,
        .rx_retries = 20
    };
    err = atcab_init(&atecc608);
```
Here is the ```i2c_master_init()``` function:
```
esp_err_t i2c_master_init()
{
    int i2c_master_port = 0;  
    i2c_config_t conf_master;
    conf_master.mode = I2C_MODE_MASTER;
    conf_master.sda_io_num = SDA_GPIO;
    conf_master.sda_pullup_en = GPIO_PULLUP_ENABLE;
    conf_master.scl_io_num = SCL_GPIO;
    conf_master.scl_pullup_en = GPIO_PULLUP_ENABLE;
    conf_master.master.clk_speed = 100000;
    i2c_param_config(i2c_master_port, &conf_master);
    return i2c_driver_install(i2c_master_port, conf_master.mode, 0, 0, 0);
}
```
I can't see in my code where I configured something wrong unless I am missing something here
Any response or help is appreciated

bryan-hunt, 2021-03-26:
There is an error in the ESP32 HAL currently. You have to revert to library version v3.1 in order to work with the esp32. It's in the list of items to fix. See #175

closed 2021-03-26 by bryan-hunt

MrMelon1232, 2021-03-26:
Hi @bryan-hunt, when was the version 3.2 released ? I am currently using the aws freertos port of this library and the last update they made on there cryptoauthlib library is in Feb 2020. I would assume they are using the v3.1 if 3.2 is not working.