MicrochipTech/cryptoauthlib #215: Using atcacert_decode_pem_cert to convert certificate from PEM to DER format fails

MrMelon1232, 2021-03-30:
Hi,

I am currently trying to write my own certificate to the atecc608a chip by converting it first to DER format from pem. Whenever I try to use the function ```atcacert_decode_pem_cert```, it fails. 
Here is my code:
```
void run_atecc_task()
{
    ATCA_STATUS err;
   
    ATCAIfaceCfg atecc608;
    err = atecc_init(&atecc608);
    printf("atecc_init() function return with status: %d\n", err);
    
    uint8_t config_zone[128];

	err = atcab_read_config_zone(config_zone);
        printf("atcab_read_config_zone() function return with status: %d\n", err);
	printf("Config zone:\n");
	for(int i = 0; i < 127; i++) printf("%u, ", config_zone[i]);
	printf("%u\n", config_zone[127]);
   
    uint8_t * dercert;
    size_t * dercert_size;
    err = atcacert_decode_pem_cert(CERT_TEST, sizeof(CERT_TEST), &dercert, &dercert_size);
    printf("atcacert_decode_pem_cert() function return with status: %d\n", err);
    printf("Encoded PEM cert in DER format: %d\n", dercert);
    printf("Sizeof der cert: %d\n", dercert_size);
    //atcacert_write_cert(atcacert, cert, sizeof(CERT_TEST));
}
```
Here is the formatting of the definition CERT_TEST
```
#define CERT_TEST   \
"-----BEGIN CERTIFICATE-----\n" \                                         
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"...............\n"    \    
"-----END CERTIFICATE-----\n"
```
Any suggestions on a different approach of what I am doing wrong here ?

MrMelon1232, 2021-03-30:
Hi @bryan-hunt,

Your solution worked I was able to encode my certificate. Thank you for the quick response it is much appreciated! I will be closing this issue.

closed 2021-03-30 by MrMelon1232

feizaipp, 2022-03-07:
Hi @MrMelon1232 ,

    I want to know how to use atcacert_write_cert API to  write my own certificate to device.
    Looking forward to your reply.
    Thanks.