MicrochipTech/cryptoauthlib #223: wrong type cast in lib/pkcs11/pkcs11_config.c

brownmonkey, 2021-05-10:
**Describe the bug**
In lib/pkcs11/pkcs11_config.c (branch main):
 `slot_ctx->interface_config.atcai2c.baud = (uint8_t)strtol(argv[3], NULL, 10);`
should be:
 `slot_ctx->interface_config.atcai2c.baud = (uint32_t)strtol(argv[3], NULL, 10);`

bryan-hunt, 2021-05-10:
So yes this cast is wrong and it'll get fixed in the next release. But keep in mind that it's actually keeping the user from getting an error when they use a value over 100k. Given there is no i2c hal for windows and generally you only use the config file with a system that has a filesystem this ends up using the linux i2c hal.

The i2c clock is controled by devicetree in linux and there is no userspace API available to change it. The only way to change it is to directly write to your processor's dividers. So yes in theory someone can write their own hal for linux that is able to change the clock frequency - but for all others a value over 100k in >v3.3.0 will cause an error. Previous versions did not return an error - they simply ignored the configuration.

brownmonkey, 2021-05-12:
> So yes in theory someone can write their own hal for linux that is able to change the clock frequency - but for all others a value over 100k in >v3.3.0 will cause an error. 

I wonder if I understood it right:
A value over 100k in >v3.3.0 will cause an error because the switching to 100k within the wakeup function does not work (because there is no userspace API available to change i2c baud)?

Does the "second_byte" in calib_wakeup_i2c() added in https://github.com/MicrochipTech/cryptoauthlib/commit/aa8024df4bebd3ceed96d0aa0954743a01fa4633
an alternative to make it works over 100k? This did not work for my linux i2c. I assume the master i2c does not proceed sending the second_byte when the address (0x00) is not acknowledged by any clients.

bryan-hunt, 2021-05-12:
> I wonder if I understood it right:
> A value over 100k in >v3.3.0 will cause an error because the switching to 100k within the wakeup function does not work (because there is no userspace API available to change i2c baud).

Correct - the only knowledge of the user intention is the configuration structure. In the previous versions the library simply ignored the value from the user and proceeded with the assumption that the user set the i2c bus to 100k or lower in devicetree. Just modify your specific configuration to make your boards usage (in linux set the baud to 100k).

> Is the "second_byte" in calib_wakeup_i2c() added in [aa8024d](https://github.com/MicrochipTech/cryptoauthlib/commit/aa8024df4bebd3ceed96d0aa0954743a01fa4633)
> an alternative to make it works over 100k? This did not work for my linux i2c. I assume my master i2c does not proceed sending the second_byte when the address (0x00) is not acknowledged by any clients.

The write to address 00 at 100k is the part that meets the pulse width requirements of the cryptoauth devices. The second byte is not required for the device communication and the expectation under normal circumstances is that it does not get transmitted. This is also why the resulting error of the i2c transaction is ignored. 

The adjustment made to add a specific second byte was to resolve several other i2c communication conditions: a) not all i2c implementations allow a zero length message, b) if you enabled i2c debugging in the linux kernel cryptoauthlib would generate a significant volume of zero length debug messages in the kernel logs, c) a write to address 00 is a special i2c transaction known as a general call, while a zero length general call is an error under all conditions, in order to address a & b the value added has to also not interfere with devices that do understand the general call, the lsb being set in that byte indicates the general call is for another i2c master on the bus - which is exceedingly rare.

brownmonkey, 2021-05-18:
So I understood:
The second byte is to make the 2-byte transmission conform to / go around:
- i2c implementations which do not allow zero length message
- kernel i2c debugging which generates significant volume zero length debug messages
- general call
=> The second byte is not part of reaching the pulse width requirements at over 100k.

How to meet the pulse width requirements at over 100k?

bryan-hunt, 2021-05-18:
> How to meet the pulse width requirements at over 100k?

You have two options - both of which will involve customizing the hal to match your platform.

* Drive SDA manually as a GPIO - if you elect to do this you will need to make a wake up function accessible via ATCA_HAL_CONTROL_WAKE in the hal control function.
* Modify the I2C clock dividers manually - if you elect to do this you will need to make a baud rate change function accessible via ATCA_HAL_CHANGE_BAUD in the hal control function.