MicrochipTech/cryptoauthlib #226: CommunicationError exception being thrown

seanpmaddox, 2021-05-18:
**Describe the bug**
I am using the ATECC608A on a custom PCB which includes an RPi CM3+, the ATECC608A sits on I2C bus 1.  I am using cryptoauthlib from python using the Microchip cryptoauthlib python wrapper package.  I had been using cryptoauthlib-20210121, this package was working fine.  I moved to a 2nd system and installed cryptoauthlib (sudo pip3 install cryptoauthlib) with no issue, this installed cryptoauthlib-20210514.  When I began testing my code (which works fine under cryptoauthlib-20210121) the CommunicationError exception is being thrown as soon as I init the stack.  To validate this is tied to this specific package I upgraded a system that was previously working on cryptoauthlib-20210121 to cryptoauthlib-20210514 via 'sudo pip3 install -U cryptoauthlib' and can confirm this system is now also receiving the CommunicationError exception.

We will be moving to the ATECC608B for production, our proto boards were built with the A.

**To Reproduce**
Steps to reproduce the behavior:
1. upgrade a system running cryptoauthlib-20210121 to cryptoauthlib-20210514
2. setup config accordingly (crypto.cfg_ateccx08a_i2c_default() in my case)
3. set cfg.atcai2c.bus = 1
4. load_cryptoauthlib()
5. result = atcab_init(config_here)
6. check result.  result in my case is 0xF0 which maps to the CommunicationError exception.

**Expected behavior**
Expect to be able to initialize the ATECC608A and perform read, write and other operations with it via the atcab python module.

**Additional context**
No additional context to provide.

bryan-hunt, 2021-05-18:
set cfg.atcai2c.baud = 100000

seanpmaddox, 2021-05-18:
Thanks Bryan - that did indeed fix it.  I've been of the impression that the I2C bus is running at 400K, am i incorrect or am i forcing the rest of my bus to be 100K via this?  I do have other chips on the same bus.  thx - sean

bryan-hunt, 2021-05-18:
See #223 for more discussion on i2c baud rate setting in linux - in short there isn't a userspace API available to change baud rate so it runs at whatever you have configured in devicetree. The difference you're seeing is due to improvements to the underlying hal architecture which is no longer ignoring the baud rate request.

seanpmaddox, 2021-05-18:
will do, thanks again Bryan.

closed 2021-05-18 by seanpmaddox