MicrochipTech/cryptoauthlib #229: cfg_ateccx08a_kithid_default Segmentation fault

pavithran93, 2021-05-31:
Hi,
         I am trying to integrate the security chip  ATECC608A with my python project. I connected the chip with my device and when I run the python code as below I got segmentation fault at the line 
         assert ATCA_SUCCESS == atcab_init(cfg_ateccx08a_kithid_default())
[image]

pavithran93, 2021-06-01:
Solved this by changing cfg_ateccx08a_kitcdc_default() to cfg_ateccx08a_i2c_default().
Now I can connect with the ATECC608A  chip.

Another question. When I request public key from the slot 0 using 
assert ATCA_SUCCESS == atcab_get_pubkey(0, publickey)
print('Public Key')
print(''.join(['%02X ' % x for x in publickey]))
print(publickey)

I am getting 64 bit byte array as below

[image]

How I can convert to public key file like below?

[image]

bryan-hunt, 2021-06-01:
For product support please submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2021-06-01 by bryan-hunt