MicrochipTech/cryptoauthlib #230: Test application can't access to ATECC608A device via I2C [Embedded Linux]

victorK8, 2021-05-31:
**Test application can't access to ATECC608A device via I2C the bug**

## Used platform

- Microchip sam5d27-wlsom1 with custom buildroot's linux distribution.
- ATECC608A with UDFN socket with XPRO connector.

##  Compiling

We cross-compile cryptoauth library and test app on Ubuntu 20.04 LTS machine (without errors or warnings).

After that, we send  all compiled content to platform via scp.

## Bug

In sam5d27-wlsom1 platform, we run the test app succesfully.

However, when we are running  `info` attribute, test app returns the next lines:

/home/victor/cryptoauthlib/lib/atca_iface.c:98:e1:Failed to configure HAL
/home/victor/cryptoauthlib/lib/atca_iface.c:54:e1:atinit
atcab_init() failed with ret=0x000000E1

ATECC608A is correctly connected to platform via I2C in XPRO connector. Running `i2cdetect -y 1` command, we can check that ATECC608A has 0x60 address in i2c bus. Result of i2cdetect command:

00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

Any idea?

Thank guys!

bryan-hunt, 2021-06-01:
Try to set the communication parameters manually:

```
info -i i2c 1 -a c0
```

Subsequent commands will use the cached parameters

bryan-hunt, 2021-06-21:
I have confirmed in v3.3.1 when the baud rate is changed to 100khz everything is working properly on a raspberry pi. The latest version that has been release v3.3.2 changed the default configuration on linux to 100khz so it should exhibit the same behavior as v3.2x

bryan-hunt, 2021-06-21:
I believe everything related to this configuration issue is resolved in v3.3.2 and the corresponding python release 20210620 where the baud rate in the default structures is set to 100khz on linux platforms.

victorK8, 2021-06-22:
Hi @bryan-hunt, 

Thanks for answering.

I'll try to set baudrate to 100kHz.

Thanks!

closed 2021-06-29 by bryan-hunt

kiranjenex, 2022-03-31:
Hi bryan-hunt,

Facing the same issue and the baud rate is set to 100000Hz. i2c-detect works properly like
 sudo i2cdetect -r -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- 35 36 -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

but when I run the test app with command 
**./cryptoauth_test info -d ecc608 -i i2c 1 -a 6C**

/home/revofi/cryptoauthlib/lib/calib/calib_read.c:103:f0:calib_read_zone - execution failed
/home/revofi/cryptoauthlib/lib/calib/calib_read.c:619:f0:calib_read_zone - falied
atcab_init() failed with ret=0x000000F0

it displays the error message I debug the code and I found that calib_wake returns the value 0xF0. which means i2c device did not wake up but the i2cdetect works which is confusing.

Please help me out on this issue.
Thanks,
Ravi

timliang73, 2023-03-06:
Did you find the root cause? thanks

victorK8, 2023-03-06:
Hi,

Yeah. It's ok now.

Thanks