MicrochipTech/cryptoauthlib #234: atcab_genkey not working

Hoshi1807, 2021-06-16:
Hi,

I'm working with the ATECC608A over i2c (i2c-dev) on a custom Linux Board.
Everything like atcab_privwrite() , atcab_ecdh() and so on are working properly...

The only thing which isn't is atcab_genkey()... i always get the Error ATCA_WAKE_SUCCESS if i call this function, no matter what i do. I tested several chips and get always this error.
I don't understand whats the meaning of ATCA_WAKE_SUCCESS and how can i avoid it ?!

Is this a bug or can you give me a hint whats the problem? I tried cryptoauthlib v3.2.3 and 3.3.1 with the same result..

greetings,

Holger

bryan-hunt, 2021-06-16:
Please provide the complete hex data for the config zone of the part in question, and the parameters you're using the in the genkey call.

Hoshi1807, 2021-06-17:
Hi Bryan-hunt,

we got the problem solved. It was a hardware issue at our layout, so the chip got only 2,46 Volts (instead of 3.3) and the current collapsed every time at complex operations like genkey. 

Now the chip get his 3.3V and everything is working as expected. 
Thank you for your answer and have a nice day.

closed 2021-06-17 by Hoshi1807