MicrochipTech/cryptoauthlib #235: ATECCx08 engine for OpenSSL1.1.1

Saadharta, 2021-06-16:
Good afternoon;

I am trying to figure out how to create an engine for an ATECC508A chip based on what has been done in the cryptoauth-openssl-engine for openSSL1.0.2.

I started by working on the EDCH / ECDSA components.... however i am having some issues thanks to the changes made in openSSL1.1.1.
(one of them being how to populate the EC_KEY_METHOD required by ENGINE_set_EC() function then initializing an ecdh-like EC_KEY and an ecdsa-like EC_key from it)

I noticed few comments in the cryptoauth-openssl-engine refering to a potential openSSL1.1.x engine implementation. 

If such a feature is still planned, could I have some advice on how to set it up?
Else, which alternatives to openSSL do I have for an SSL / TLS communication using the ATECC508A abilities?

Thanks in advance, 
Saad

bryan-hunt, 2021-06-16:
There is absolutely no reason to do so and simply results in endless frustration with the OpenSSL api changing all the time. There is already a functioning engine that works for all versions of OpenSSL with a very large support base. That is the pkcs11 engine (i.e. [libp11](https://github.com/OpenSC/libp11) ) that is already available in virtually all linux distributions.

Instructions for using cryptoauthlib as a pkcs11 provider can be found on the cryptoauthlib wiki: https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11

In summary - writing your own OpenSSL engine is a hugely bad idea.

closed 2021-06-16 by bryan-hunt

Saadharta, 2021-06-16:
Thank you for your fast answer (and helping me keeping my sanity);

I was focused on openSSL alone and missed the PKCS11 engine, which I thought was something unrelated (my target is a Raspberry Pi).