MicrochipTech/cryptoauthlib #237: Raspberry Pi Zero keeps giving the same error

CrabbyPete, 2021-06-24:
I have an ATECC608 A connected to the I2c bus of a Raspberry Pi Zero. I can see the device with i2cdetect. No matter which example I run I get the following:

Traceback (most recent call last):
  File "create_csr.py", line 172, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "create_csr.py", line 95, in info
    cfg = eval('cfg_at{}a_{}_default()'.format(atca_names_map.get(device), atca_names_map.get(iface)))
  File "<string>", line 1, in <module>
  File "/home/pi/atecc608/lib/python3.7/site-packages/cryptoauthlib/iface.py", line 165, in cfg_ateccx08a_i2c_default
    return _iface_load_default_config('cfg_ateccx08a_i2c_default')
  File "/home/pi/atecc608/lib/python3.7/site-packages/cryptoauthlib/iface.py", line 160, in _iface_load_default_config
    raise UnsupportedInterface('The selected interface specified by {} is not supported by this version of the library'.format(name))
cryptoauthlib.exceptions.UnsupportedInterface: The selected interface specified by cfg_ateccx08a_i2c_default is not supported by this version of the library

bryan-hunt, 2021-06-24:
Do you have libcryptoauth.so installed in your system libraries? If you have something installed in your system path the expectation of the python module is that you intended for it to be appropriate for your system.

CrabbyPete, 2021-06-25:
Thanks for the quick reply. That was it.

closed 2021-06-25 by CrabbyPete

dexterac, 2022-02-09:
I'm running into similar issue with info.py

My setup:
1. Raspberry Pi 4b
2. Raspbian GNU/Linux 11 (bullseye)
3. armv7l (32bit)
4. ATECC608A : https://www.adafruit.com/product/4314

`python cryptoauthtools/python/examples/info.py -i i2c
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/cryptoauthlib/iface.py", line 160, in _iface_load_default_config
    return ATCAIfaceCfg.in_dll(get_cryptoauthlib(), name)
ValueError: /lib/arm-linux-gnueabihf/libcryptoauth.so: undefined symbol: cfg_ateccx08a_i2c_default

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/cryptoauthlib/python/cryptoauthtools/python/examples/info.py", line 93, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "/home/pi/cryptoauthlib/python/cryptoauthtools/python/examples/info.py", line 32, in info
    cfg = eval('cfg_at{}a_{}_default()'.format(atca_names_map.get(device), atca_names_map.get(iface)))
  File "<string>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/cryptoauthlib/iface.py", line 167, in cfg_ateccx08a_i2c_default
    return _iface_load_default_config('cfg_ateccx08a_i2c_default')
  File "/home/pi/.local/lib/python3.9/site-packages/cryptoauthlib/iface.py", line 162, in _iface_load_default_config
    raise UnsupportedInterface('The selected interface specified by {} is not supported by this version of the library'.format(name))
cryptoauthlib.exceptions.UnsupportedInterface: The selected interface specified by cfg_ateccx08a_i2c_default is not supported by this version of the library`