MicrochipTech/cryptoauthlib #241: v3.3.2: p11tool --list-all does not see the available objects in token

ambarus, 2021-06-30:
**Describe the bug**
root@sama5d27-wlsom1-ek-sd:~# cat /etc/cryptoauthlib/cryptoauthlib.conf
# Cryptoauthlib Configuration File

filestore = /var/lib/cryptoauthlib

root@sama5d27-wlsom1-ek-sd:~# cat /var/lib/cryptoauthlib/0.conf 
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

# Set a label for this slot (optional) - will default to <slot>ABC so
# 0.conf will have a default label 00ABC
#label = MCHP

# Configure the device interface for an enabled HAL
# hid,i2c,<address>
# i2c,<address>,<bus>
# spi,<select_line>,<baud>
interface = i2c,0x6A,0

# Configure the device type - base part number (optional)
device = ATECC608A-TFLXTLS

#Configure open slots for additional pkcs11 objects (optional)
#freeslots = 1,2,3

# Manually configure keys into device locations (slots/handles)

# Slot 0 is the primary private key
#object = private,device,0

# Slot 15 is a public key
#object = public,root,15
root@sama5d27-wlsom1-ek-sd:~# 

root@sama5d27-wlsom1-ek-sd:~# cat /usr/share/p11-kit/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no

root@sama5d27-wlsom1-ek-sd:~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- 35 -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  

root@sama5d27-wlsom1-ek-sd:~# p11tool --list-all
pkcs11_init: PKCS #11 initialization error.
warning: no token URL was provided for this operation; the available tokens are:

**To Reproduce**
Steps to reproduce the behavior:
linux4sam: ATECC608 (the one populated on sama5d27-wlsom1-ek) using cryptoauthlib v3.3.2. 

**Expected behavior**
List available tokens.

**Additional context**
Same configuration but with cryptoauthlib v3.2.5, works just fine.

bryan-hunt, 2021-06-30:
Can you run ldd on the resulting libcryptoauth.so while on the target?

ambarus, 2021-07-01:
root@sama5d27-wlsom1-ek-sd:~# ldd /usr/lib/libcryptoauth.so
	libcrypto.so.1.1 => /usr/lib/libcrypto.so.1.1 (0xb6db0000)
	librt.so.1 => /lib/librt.so.1 (0xb6d9a000)
	libc.so.6 => /lib/libc.so.6 (0xb6ca8000)
	/lib/ld-linux-armhf.so.3 (0xb6f98000)
	libdl.so.2 => /lib/libdl.so.2 (0xb6c95000)
	libpthread.so.0 => /lib/libpthread.so.0 (0xb6c70000)
root@sama5d27-wlsom1-ek-sd:~# ldd /usr/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.so
	libc.so.6 => /lib/libc.so.6 (0xb6dc7000)
	/lib/ld-linux-armhf.so.3 (0xb6edd000)
root@sama5d27-wlsom1-ek-sd:~# 

The following YP recipe conf may help. Is the EXTRA_OECMAKE conf ok for v3.3.2?
```
SRCREV = "2dc8384e752f3f9154019b1af1bb8734d61e9f9f"

S = "${WORKDIR}/git"

DEPENDS = "udev openssl"
RDEPENDS_${PN} = "libp11 (>= 0.4.10) gnutls-bin"
RRECOMMENDS_${PN} = "p11-kit"

inherit cmake

EXTRA_OECMAKE = " \
    -DATCA_HAL_I2C=ON \
    -DATCA_PKCS11=ON \
    -DATCA_OPENSSL=ON \
    -DATCA_ATECC508A_SUPPORT=ON \
    -DATCA_ATECC608_SUPPORT=ON \
    -DATCA_BUILD_SHARED_LIBS=ON \
    -DATCA_TNGTLS_SUPPORT=ON \
    -DATCA_TNGLORA_SUPPORT=ON \
    -DATCA_TFLEX_SUPPORT=ON \
    -DATCA_USE_ATCAB_FUNCTIONS=ON \
"

CFLAGS += "-fcommon"

```

bryan-hunt, 2021-07-01:
I'm not able to replicate this failure. I'm seeing the expected behavior in ubuntu 20 LTS and Raspbian (Buster). The library should no longer throw errors when compiled with -fno-common per gcc 10 defaults so you should be able to remove the -fcommon flag.

If you have p11-kit installed can you turn on call logging to see if it is able to open the library and successfully call the api?

ambarus, 2021-07-01:
Do you mean setting "log-calls: yes" in /usr/share/p11-kit/modules/cryptoauthlib.module?
```
root@sama5d27-wlsom1-ek-sd:~# cat /etc/pkcs11/pkcs11.conf
user-config: merge
root@sama5d27-wlsom1-ek-sd:~# cat /usr/share/p11-kit/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: yes
root@sama5d27-wlsom1-ek-sd:~# p11tool --list-all                                 
pkcs11_init: PKCS #11 initialization error.
warning: no token URL was provided for this operation; the available tokens are:

root@sama5d27-wlsom1-ek-sd:~# 
```

bryan-hunt, 2021-07-01:
It looks like something is wrong with the built library. Can you dump the symbols?

bryan-hunt, 2021-07-01:
Try building with this patch applied:

```
diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index bdc7891f..6dbaea0d 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -247,7 +247,7 @@ set(USE_UDEV TRUE)
 elseif(LIBUSB_GOOD AND NOT ATCA_HAL_KIT_CDC)
 set(USE_LIBUSB TRUE)
 else()
-message(FATAL_ERROR, "Missing Build Dependencies for USB - install libusb-1.0-0-dev or libudev-dev")
+message(FATAL_ERROR "Missing Build Dependencies for USB - install libusb-1.0-0-dev or libudev-dev")
 endif()

 endif(LINUX AND NEED_USB)

```

ambarus, 2021-07-02:
NEED_USB is false, so your message is never hit. I've tried:
```
diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index bdc7891f8186..0974b45b10ba 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -247,9 +247,11 @@ set(USE_UDEV TRUE)
 elseif(LIBUSB_GOOD AND NOT ATCA_HAL_KIT_CDC)
 set(USE_LIBUSB TRUE)
 else()
-message(FATAL_ERROR, "Missing Build Dependencies for USB - install libusb-1.0-0-dev or libudev-dev")
+message(FATAL_ERROR "Missing Build Dependencies for USB - install libusb-1.0-0-dev or libudev-dev")
 endif()
 
+else()
+message(FATAL_ERROR "NEED_USB is false")
 endif(LINUX AND NEED_USB)
 
 if(USE_UDEV)
```

and I got:
```
ERROR: cryptoauthlib-1.0+git999-r0 do_configure: Execution of '/home/tudor/work/yocto-mchp/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+git999-r0/temp/run.do_configure.24400' failed with exit code 1:
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /home/tudor/work/yocto-mchp/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+git999-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc
-- Check for working C compiler: /home/tudor/work/yocto-mchp/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+git999-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for malloc
-- Looking for malloc - found
-- Looking for free
-- Looking for free - found
-- Found OpenSSL: /home/tudor/work/yocto-mchp/poky/build-microchip/tmp/work/cortexa5t2hf-neon-vfpv4-poky-linux-gnueabi/cryptoauthlib/1.0+git999-r0/recipe-sysroot/usr/lib/libcrypto.so (found version "1.1.1k")  
CMake Error at lib/CMakeLists.txt:254 (message):
  NEED_USB is false
```
I'll check how I can dump the library symbols. I don't know how.

ambarus, 2021-07-02:
`root@sama5d27-wlsom1-ek-sd:~# objdump -T /usr/lib/libcryptoauth.so

/usr/lib/libcryptoauth.so:     file format elf32-littlearm

DYNAMIC SYMBOL TABLE:
0000c638 l    d  .init	00000000              .init
00035000 l    d  .data	00000000              .data
00000000      DF *UND*	00000000  OPENSSL_1_1_1 ECDSA_SIG_get0_r
00000000      DF *UND*	00000000  GLIBC_2.4   open
00000000      DF *UND*	00000000  OPENSSL_1_1_0 HMAC_CTX_new
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_DigestInit_ex
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_sign
00000000      DF *UND*	00000000  OPENSSL_1_1_0 CMAC_Init
00000000      DF *UND*	00000000  GLIBC_2.4   __ctype_toupper_loc
00000000      DF *UND*	00000000  OPENSSL_1_1_0 CMAC_CTX_new
00000000      DF *UND*	00000000  GLIBC_2.4   mmap
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EC_KEY_free
00000000      DF *UND*	00000000  OPENSSL_1_1_1 RSA_pkey_ctx_ctrl
00000000      DF *UND*	00000000  GLIBC_2.4   memcmp
00000000  w   D  *UND*	00000000              __gmon_start__
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_MD_CTX_reset
00000000      D  *UND*	00000000              pthread_mutexattr_settype
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_CTX_new
00000000      DF *UND*	00000000  GLIBC_2.4   strncpy
00000000      DF *UND*	00000000  GLIBC_2.4   fclose
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_new
00000000      DF *UND*	00000000  GLIBC_2.4   strchr
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_get0_EC_KEY
00000000      DF *UND*	00000000  OPENSSL_1_1_0 HMAC_Update
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EC_KEY_get0_public_key
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_sign_init
00000000      DF *UND*	00000000  GLIBC_2.4   calloc
00000000      DF *UND*	00000000  GLIBC_2.4   fopen
00000000      DF *UND*	00000000  OPENSSL_1_1_0 BN_bin2bn
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_CIPHER_CTX_ctrl
00000000      DF *UND*	00000000  GLIBC_2.4   memset
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_DecryptInit_ex
00000000      D  *UND*	00000000              pthread_mutexattr_setrobust
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EC_POINT_new
00000000      D  *UND*	00000000              pthread_mutexattr_setpshared
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_id
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EC_KEY_set_public_key
00000000      DF *UND*	00000000  OPENSSL_1_1_0 CMAC_Update
00000000      DF *UND*	00000000  OPENSSL_1_1_0 CMAC_Final
00000000  w   D  *UND*	00000000              _ITM_deregisterTMCloneTable
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_DigestUpdate
00000000      DF *UND*	00000000  GLIBC_2.4   usleep
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EC_KEY_get0_group
00000000      DF *UND*	00000000  GLIBC_2.4   free
00000000      DF *UND*	00000000  GLIBC_2.4   read
00000000      DF *UND*	00000000  GLIBC_2.4   write
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_DigestFinal_ex
00000000  w   D  *UND*	00000000              _ITM_registerTMCloneTable
00000000      DF *UND*	00000000  GLIBC_2.4   fseek
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EVP_PKEY_set1_EC_KEY
00000000      DF *UND*	00000000  GLIBC_2.4   __memcpy_chk
00000000      DF *UND*	00000000  GLIBC_2.4   pthread_mutex_unlock
00000000      DF *UND*	00000000  OPENSSL_1_1_0 ECDSA_do_verify
00000000      DF *UND*	00000000  GLIBC_2.4   ioctl
00000000      DF *UND*	00000000  OPENSSL_1_1_0 ECDSA_SIG_new
00000000      D  *UND*	00000000              atcac_sw_random
00000000      DF *UND*	00000000  GLIBC_2.4   pthread_mutex_lock
00000000      DF *UND*	00000000  GLIBC_2.4   strlen
00000000      DF *UND*	00000000  OPENSSL_1_1_0 PEM_read_bio_PrivateKey
00000000      DF *UND*	00000000  GLIBC_2.4   shm_open
00000000      DF *UND*	00000000  OPENSSL_1_1_0 EC_KEY_new_by_curve_name
00000000      DF *UND*	00000000  GLIBC_2.4   memcpy
00000000      DF *UND*	00000000  OPENSSL_1_1_0 i2o_ECPublicKey
00000000      DF *UND*	00000000  OPENSSL_1_1_1 ECDSA_SIG_get0_s
00000000      DF *UND*	00000000  GLIBC_2.4   __ctype_tolower_loc
[... 848 lines cut ...]
0001adb8 g    DF .text	00000046  Base        atcab_aes_cbc_init_ext
00017b74 g    DF .text	0000001e  Base        atcacert_get_response
0001af9c g    DF .text	0000001a  Base        atcab_aes_cbcmac_init
00020f94 g    DF .text	00000020  Base        pkcs11_os_create_mutex
0000ec28 g    DF .text	00000058  Base        atcab_ecdh_base
00011ea4 g    DF .text	00000014  Base        atInfo
0000f390 g    DF .text	00000040  Base        atcab_nonce_rand
00013470 g    DF .text	00000002  Base        calib_challenge_seed_update
000163f8 g    DF .text	000001e8  Base        atcah_check_mac
0001ca7c g    DF .text	00000064  Base        tng_atcacert_max_device_cert_size
0000f1d4 g    DF .text	0000003c  Base        atcab_lock_data_zone_crc
0000fef0 g    DF .text	00000024  Base        atcab_sign
0000f4d4 g    DF .text	00000018  Base        atcab_random
0000fbf8 g    DF .text	00000048  Base        atcab_hw_sha2_256
0001bc4c g    DF .text	00000134  Base        atcab_pbkdf2_sha256_ext
0002092c g    DF .text	00000004  Base        C_WrapKey
0000f19c g    DF .text	00000038  Base        atcab_lock_data_zone
0001d0d0 g    DF .text	00000074  Base        atcac_aes_gcm_encrypt_start
0000ff14 g    DF .text	00000058  Base        atcab_sign_internal
00010fe8 g    DF .text	00000004  Base        atca_gen_key_in_out_t_size
00013404 g    DF .text	00000016  Base        calib_nonce
00013174 g    DF .text	00000008  Base        calib_lock_data_zone
00010528 g    DF .text	0000000a  Base        releaseATCADevice

`

ambarus, 2021-07-06:
Any other suggestions on how to tackle this?

Saadharta, 2021-07-06:
can you test a simple p11-kit list-modules -v????

bryan-hunt, 2021-07-06:
@ambarus Disable OpenSSL - it serves no purpose in this configuration.

bryan-hunt, 2021-07-06:
@ambarus Okay stick with v3.2.5 or wait for v3.3.3 before updating the recipes.

ambarus, 2021-07-06:
> @ambarus Disable the OpenSSL option in your build file - it serves no purpose in this configuration.

I disabled openssl, now p11tool sees the objects, but I get a segmentation fault:
```
root@sama5d27-wlsom1-ek-sd:~# p11tool --list-all
Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=237C30CB6AC65A01;token=00ABC;object=device;type=private
	Type: Private key (EC/ECDSA-SECP256R1)
	Label: device
	Flags: CKA_PRIVATE; CKA_SENSITIVE; 
	ID: 

Object 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=237C30CB6AC65A01;token=00ABC;object=device;type=public
	Type: Public key (EC/ECDSA-SECP256R1)
	Label: device
	ID: 

Object 2:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=237C30CB6AC65A01;token=00ABC;object=device;type=cert
	Type: X.509 Certificate
	Label: device
	ID: 

Object 3:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=237C30CB6AC65A01;token=00ABC;object=signer;type=cert
	Type: X.509 Certificate
	Label: signer
	Flags: CKA_CERTIFICATE_CATEGORY=CA; CKA_TRUSTED; 
	ID: 

Segmentation fault
root@sama5d27-wlsom1-ek-sd:~# 

```

bryan-hunt, 2021-07-06:
Yes - #240 documents that. Because v3.3.3 is coming out next week that is why I suggest sticking with v3.2.5 or wait.