MicrochipTech/cryptoauthlib #242: Error with hal_sam0_i2c_asf.c in wlr089 chip (ARM Cortex M0)

victorK8, 2021-07-02:
**Cryptoauth library doesn't work **

HI,

I'm trying to add ATECC608 in my LoRa module based on WLR089 chip (ARM Cortex M0). ATECC608 and WLR089 are connected via i2c.

I use C language in Microchip Studio for developing the Firmware of WLR089.

Working with cryptoauth library, I watched errors in calib_read_zone functions. By debugging, I concluded that hal i2c interface doesn't work well.

Moreover, there are some thing that seems weird in the code (We add a fragment of it). For example, the line `plib = (i2c_sam0_instance_t*)cfg->cfg_data;`.  They are different structs.

ATCA_STATUS hal_i2c_send(ATCAIface iface, uint8_t word_address, uint8_t *txdata, int txlength)
{
    ATCAIfaceCfg *cfg = atgetifacecfg(iface);
    struct i2c_master_module * i2c_master_instance;
    i2c_sam0_instance_t * plib;

    struct i2c_master_packet packet = {
        .address            = cfg->atcai2c.address >> 1,
        .data_length        = txlength,
        .data               = txdata,
        .ten_bit_address    = false,
        .high_speed         = false,
        .hs_master_code     = 0x0,
    };

    if (!cfg)
    {
        return ATCA_BAD_PARAM;
    }

   plib = (i2c_sam0_instance_t*)cfg->cfg_data;

Have anybody a bug like this? Any help?

thanks!

bryan-hunt, 2021-07-02:
You're going to have to more carefully describe the bug. The issue template might assist you in that. What is the behavior that you are seeing? What is the actual error?

You've pasted a copy of a packet while referencing hal configuration data. The configuration data for an I2C device stays the same - the packet is specific to data being sent.

victorK8, 2021-07-05:
Hi,

I have the next error :
```
../src/cryptolib/calib/calib_read.c:103:e2:calib_read_zone - execution failed
../src/cryptolib/calib/calib_read.c:619:e2:calib_read_zone - falied
```

My i2c ondifguration is (I try with 40000 and 10000 baudrate):
```
	
	ATCAIfaceCfg iface_conf = {
		  .iface_type            = ATCA_I2C_IFACE,
		  .devtype               = ATECC608A,
		  {
			  .atcai2c           = {
				  .address       = 0xC0,
				  .bus           = 1,
				  .baud          = 100000, //400000,
			  },
		  },
		  .wake_delay            = 1500,
		  .rx_retries            = 20
	  };

```

Thaks! Is it needed more info?_

bryan-hunt, 2021-07-06:
Diagnosing your board in ASF is outside the support that is provided here. You need to submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2021-07-06 by bryan-hunt