MicrochipTech/cryptoauthlib #243: $ p11 --list-tokens fails and p11-kit list-modules -v  return "/usr/lib/libcryptoauth.so: undefined symbol: atcac_sw_random"

Saadharta, 2021-07-06:
Good day Mr Hunt; 

**Describe the bug**
I am trying to set up an ATECC508A for a TLS session via the PKCS11 engine. 

My configuration: 
    Rapsberry Pi 3;
    Fresh Raspios Buster Lite (release of 2021-05-07); 
    Kernel version 5.10.17; 
    Libssl-dev installed to get access to ssl headers;
    cryptoauthlib version 3.3.2
    An ATECC508A on I2C bus (defauld 0xC0 address).

When following the PKCS11 tutorial provided in the wiki, during testing p11tool --list-tokens --debug 9999 outputs the following:
`$ p11tool --list-tokens --debug 9999
Setting log level to 9999
|<3>| ASSERT: ../../lib/pkcs11.c[auto_load]:935
|<2>| Cannot initialize registered modules: aborting initialization because module 'cryptoauthlib' was marked as critical
|<3>| ASSERT: ../../lib/pkcs11.c[compat_load]:894
pkcs11_init: PKCS #11 initialization error.
|<3>| ASSERT: ../../lib/pkcs11.c[_pkcs11_traverse_tokens]:1647
|<3>| ASSERT: ../../lib/pkcs11.c[_gnutls_pkcs11_token_get_url]:2412
`

While trying to figure out whitch config i had botched i ended up checking
`$ p11-kit list-modules -v
p11-kit: couldn't load module: /usr/lib/libcryptoauth.so: /usr/lib/libcryptoauth.so: undefined symbol: atcac_sw_random
p11-kit: aborting initialization because module 'cryptoauthlib' was marked as critical`

**To Reproduce**

1) Follow the wiki PKCS11 installation procedure
1.1) When building the cryptoauthlib library:  cmake -D ATCA_HAL_I2C=ON -D ATCA_PKCS11=ON -D ATCA_OPENSSL=ON -D ATCA_ATECC508A_SUPPORT=ON -D ATCA_ATECC608_SUPPORT=ON -D ATCA_BUILD_SHARED_LIBS=ON -D ATCA_USE_ATCAB_FUNCTIONS=ON ./`

1.2) Edit the pkcs11/pkcs11_config.h file to enable PKCS11_508_SUPPORT and PKCS_608_SUPPORT (i am planning to upgrade to an ATECC608B once i get satisfactory results)

1.3) make
1.4) sudo make install

**Expected behavior**
p11tool --list-tokens should return a list of tokens, including my ATECC508A.

**Additional context**
ATCA_ENABLE_RAND_IMPL is only defined if ATCA_WOLFSSL and ATCA_OPENSSL and ATCA_MBEDTLS aren't defined (in _crypto/atca_crypto_sw_h_  (i fixed it on my side by adding a `#define ATCA_ENABLE_RAND_IMPL 1` in atca_crypto_sw_rand.c)

During the make process i also got this warning: 
`
[ 96%] Building C object lib/CMakeFiles/cryptoauth.dir/pkcs11/pkcs11_slot.c.o
/home/pi/cryptoauthlib/lib/pkcs11/pkcs11_slot.c: In function ‘pkcs11_slot_init’:
/home/pi/cryptoauthlib/lib/pkcs11/pkcs11_slot.c:222:22: warning: implicit declaration of function ‘pkcs11_slot_check_device_type’; did you mean ‘pkcs11_object_get_type’? [-Wimplicit-function-declaration]
             status = pkcs11_slot_check_device_type(ifacecfg);
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                      pkcs11_object_get_type
` 
God bless the Verbose options. 
` $ p11-kit list-modules -v
p11-kit: couldn't load module: /usr/lib/libcryptoauth.so: /usr/lib/libcryptoauth.so: undefined symbol: pkcs11_slot_check_device_type` 

If I add a dummy function returning ATCA_SUCCESS I end up with  a Segmentation Fault : 
`C_GetTokenInfo = CKR_OK
    token: 00HSM
        manufacturer: Microchip Technology Inc
        model: ATECC508A
        serial-number: 23C11D970800D5EE
        firmware-version: 255.255
        flags:
               rng
               token-initialized
C_Finalize
  IN: pReserved = NULL
Segmentation fault`
`

bryan-hunt, 2021-07-06:
For the moment don't enable OpenSSL since your configuration doesn't use anything from it.

Saadharta, 2021-07-06:
I ran 
``` cmake -D ATCA_HAL_I2C=ON -D ATCA_PKCS11=ON -D ATCA_OPENSSL=OFF -D ATCA_ATECC508A_SUPPORT=ON -D ATCA_ATECC608_SUPPORT=ON -D ATCA_BUILD_SHARED_LIBS=ON -D ATCA_USE_ATCAB_FUNCTIONS=ON . 
```
then make / sudo make install;

Now i am running into a SegFault while using `p11-kist list-modules -v`:
```
C_Finalize
  IN: pReserved = NULL
Segmentation fault
```

closed 2021-07-06 by bryan-hunt