MicrochipTech/cryptoauthlib #245: HMAC Message Bug

gskrish7, 2021-07-11:
Hi,

Good Day!

I am not sure, this is a bug. Giving my observation here, while running a program for HMAC Computation on atecc608a. I am trying to sign test message 0xAB*64

I am not sure is it signed correctly. But while printing the command from function `SERCOM2_I2C_Write`; results the parameter as 0x00s.; the whole sha256-HMAC command prints following, was it a bug? Does the test message for signing has to on global variable?

```034747014000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000007ed2```

From this we could see the command takes different stream of array than given one.

```
static int32_t i32apiATCAB_HMAC_Performance( void )
{
    atStatus = ATCA_SUCCESS;
    
    uint8_t pu8TestMessage[64];
    uint8_t pu8Hmac[MAC_SIZE];
    for( int i=0; i<64; pu8TestMessage[i++] = 0xAB );
    do
    {
        if( ATCA_SUCCESS != ( atStatus = atcab_init( &atecc608a_0_init_data ) ) )
        {
            printf( "atcab_init: %02x\r\n", atStatus );
            break;
        }
        atStatus = atcab_sha_hmac( pu8TestMessage, sizeof(pu8TestMessage), 5, pu8Hmac, SHA_MODE_TARGET_OUT_ONLY);
        if( ATCA_SUCCESS != atStatus )
        {
          break;
        }

    }while(0);    
    
    atcab_release();
    
    return ( (int32_t)atStatus );
}
```

```
bool SERCOM2_I2C_Write(uint16_t address, uint8_t *pdata, uint32_t length)
{
    printf( "SERCOM2_I2C_Write:%02X\r\n", address );
    for( int i=0; i<length; printf("%02x", pdata[i++] ) );
    printf( "\r\n" );
...
}
```

bryan-hunt, 2021-07-11:
See section 6.8.5.3 of the [C Standard](http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf)

The third expression of a For loop is executed only once at the end of an iteration. The first iteration of this loop the array will be as it was (all zeros).

bryan-hunt, 2021-07-11:
atcab_init for a 608 performs a read operation. You should not have the atcab_init operation inside of a loop regardless of your implementation. Only one atcab_init is ever required.

Because of #244 you might be seeing your device reset. V3.3.3 is being released this coming week to resolve the atcab_release issue.

gskrish7, 2021-07-12:
I have called `atcab_init` before `do {} while` loop, still same observation. Moreover, the while(0) would not execute more than once.

But the observation is, the wakeup is sent in between hmac sequence as follows
1. Start mode: Initiate SHA256-HMAC.
2. Send Wake-up
3. Update mode: Input 64 byte message SHA256-HMAC digest
4. Send Wake-up
5. Finalize mode: Read HMAC Digest

Thanks,

closed 2022-11-13 by github-actions[bot]