MicrochipTech/cryptoauthlib #249: ECDH key exchange between 2 ATECCx08 devices

PhillyNJ, 2021-07-28:
Hi, 

I am using the ECDH command to create a pre-master secrets from slot 1 and slot 2 on a single ATECCx08 device. My slot and key config are defined as:

slot 1 config 0x20 0x87
Key 1 config: 0x04 0xF3
slot 2 config 0x20 0x87
Ket 2 config 0x04 0xB3

I am only testing on a single device to develop my code. I am able to generate the pre-master secrets without issue. I am confused on how Bob and Alice first agree on a shared random number when using 2 ATECCx08 devices. From what I understand, the ECDH protocol starts out with a known random secret known by both Bob and Alice. If Bob is on coming from one ATECCx08 device and Alice on another, how do the 2 separate devices agree on this random number?

Thanks in advance

Phil

bryan-hunt, 2021-07-28:
ECDH creates a known secret from exchanged public keys using elliptic curve principles. It does not depend on a preshared key. 

https://en.wikipedia.org/wiki/Elliptic-curve_Diffie–Hellman

PhillyNJ, 2021-07-28:
Thank for the quick response. If I understand correctly, If an adversary was sniffing the i2c bus that exchanges the public keys, they couldn't calculate the PMS because the adversary would need to know either of the private keys, which with the ATECCx08, is impossible because the slot is locked and protected from reading (if configure as a private key)?

bryan-hunt, 2021-07-28:
Yes ECDH that is the general principle behind Diffie–Hellman key exchange.

As an aside private keys can never be read from a cryptoauth device under any circumstance.

closed 2021-07-28 by PhillyNJ