MicrochipTech/cryptoauthlib #252: NULL dereference in `calib_ecc204_sign`

sU8U7SfkcwTJVH7PjaVmej7D, 2021-08-06:
In : https://github.com/MicrochipTech/cryptoauthlib/blob/b177b31182573f439191214d356a34f5db048beb/lib/calib/calib_sign.c#L241

`memcpy` destination in `signature`, which is known to be NULL in L237.

I am not familiar with the code at all, so I did not even try to think about what the fix would be. :)

janekm, 2021-08-18:
Just noticed the same, this is causing (deserved) compiler errors for me. Like... are you guys just compiling this library with compiler warnings ignored? This is literally a cryptography library...

And no comment for 12 days?

bryan-hunt, 2021-08-19:
This is slated to be fixed in the upcoming version. This function is exclusive to a device that is not available yet so this issue has no  impact to current users.

bryan-hunt, 2021-10-08:
Resolved in v3.3.3