MicrochipTech/cryptoauthlib #255: How to transform the X and Y integers big-endian format to a base64 string valid public key format?

larinnim, 2021-08-23:
I was able to generate the private and public key. The private key is stored on Slot1 and public on slot 11. I am using atcab_get_pubkey() to return the public key. How do I transform from X and Y integers in big-endian format to base64 string valid public key format so I can put it on Google IOT cloud? I tried to use atcab_base64encode() but it's inserting weird characters and my public key is invalid on google iot core. It always add "8AhyCPIMSPj4+Pj5+Pr4+PjwAAAAAAAAAAAA=="

Reading Config Zone...
01 23 fe 83 00 00 60 02 
2a 6c e6 a8 ee c1 55 00 
60 00 00 00 8f 00 87 20 
8f 20 c4 8f 8f 8f 8f 8f 
9f 8f af 8f 8f 8f 00 00 
00 00 00 00 00 00 00 00 
00 00 af 8f ff ff ff ff 
00 00 00 00 ff ff ff ff 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
fd f7 00 00 00 00 00 00 
13 00 33 00 33 00 1c 00 
1c 00 1c 00 1c 00 1c 00 
10 00 3c 00 3c 00 3c 00 
3c 00 3c 00 3c 00 1c 00 
Device type: 3
Config Zone: LOCKED
Data Zone: LOCKED
Data Slot 1 LOCKED
Generated public key is: {X:156314630009d3cc7e84ebe3fbc630b359591780f92e5400efe11b4eb0a1c5a1, Y:9410a35145af98e6be22c0ee2ec3b383554262d46dd615cf9f152c05257dbdfb}
Data Slot 11 is LOCKED
ATECCx08 @ 0x30: rev 0x6002 S/N 0x123fe832a6ce6a8ee, zone lock status: yes, yes
Slot[0] config: 008f
Slot[1] config: 0000
Slot[2] config: 0087
Slot[3] config: 0020
Slot[4] config: 008f
Slot[5] config: 0020
Slot[6] config: 00c4
Slot[7] config: 008f
0x15, 0x63, 0x14, 0x63, 
-----BEGIN PUBLIC KEY-----
FWMUYwAJ08x+hOvj+8Yws1lZF4D5LlQA7+EbTrChxaGUEKNRRa+Y5r4iwO4uw7OD
VUJi1G3WFc+fFSwFJX29+48AhyCPIMSPj4+Pj5+Pr4+PjwAAAAAAAAAAAA==
-----END PUBLIC KEY-----

```  
const uint8_t public_key_x509_header[] = {0x30, 0x59, 0x30, 0x13, 0x06, 0x07, 0x2A, 0x86,
                                          0x48, 0xCE, 0x3D, 0x02, 0x01, 0x06, 0x08, 0x2A,
                                          0x86, 0x48, 0xCE, 0x3D, 0x03, 0x01, 0x07, 0x03,
                                          0x42, 0x00, 0x04};
char buf[128];
    size_t buf_len = sizeof(buf);
    /* Convert to base 64 */
    atcab_base64encode(public_key, ATCA_PUB_KEY_SIZE + sizeof(public_key_x509_header), buf, &buf_len);

    for (int i = 0; i < strlen((char *)public_key); i++)
    {
        printf("0x%02x, ", public_key[i]);
    }
    printf("\n");
    /* Add a null terminator */
    buf[buf_len] = 0;

    /* Print out the key */
    DEBUG_PRINTF("-----BEGIN PUBLIC KEY-----\r\n%s\r\n-----END PUBLIC KEY-----\r\n", buf);

bryan-hunt, 2021-08-23:
There are couple of ways to do this. The first is to cryptographic library and create a key object with the proper curve and give it the public key X & Y ints. However since the ASN.1 encoding for a SECP256r1 public key is known you can define the header as being static and append the X & Y ints. 

In your example above you can see the public_key_x509_header contains invalid characters - it is not yet encoded. You can see an example of this method here:

https://github.com/MicrochipTech/cryptoauthlib/blob/main/lib/pkcs11/pkcs11_key.c#L191-L229

The first step is to fill the buffer with the header, append the public key ints, then encode the complete buffer.

larinnim, 2021-08-25:
It worked! Thanks a bunch @bryan-hunt

closed 2021-08-25 by larinnim