MicrochipTech/cryptoauthlib #257: Deploying AES public Key to ATEC608B

RnScully, 2021-09-10:
The write command only writes 32 bytes, but the AES public Keys are 64 bytes. I'm confused as to how I can use the device in the field if I cannot give it the public Key of another device for ECDH encryption and decryption...Though I think I've missed something obvious. How am I supposed to load public keys into the device?

bryan-hunt, 2021-09-10:
AES keys are symmetric and are always secret - there is no public component.  If you want to store an EC public key (of curve SECP256r1) then use the atcab_write_pubkey function.

bryan-hunt, 2021-09-10:
Also ECDH is for key negotiation and not encryption/decryption. The ecdh command accepts the remote public key rather than using a stored key.

There are several examples in the test code. I would suggest reviewing the code manual as well as the datasheet for the part.

closed 2021-09-10 by bryan-hunt