MicrochipTech/cryptoauthlib #258: Possible out of bound array access in Cryptoauthlib

rspworks, 2021-09-16:
Helly Bryan, 

https://github.com/MicrochipTech/cryptoauthlib/blob/b177b31182573f439191214d356a34f5db048beb/lib/pkcs11/pkcs11_config.c#L484

In the above code, argv is a local variable and there is possible out of bound access here while doing argv[3]

Please fix

bryan-hunt, 2021-09-17:
https://github.com/MicrochipTech/cryptoauthlib/blob/b177b31182573f439191214d356a34f5db048beb/lib/pkcs11/pkcs11_config.c#L421-L426 prevents that from ever occurring.

The fourth entry in the secret key configuration will simply never get parsed.

rspworks, 2021-09-17:
Yes. Logically it may not enter that code path but code-wise it looks incorrect that the 4th element is accessed from an array of size 3.

Thanks prompt for reply

closed 2021-10-08 by bryan-hunt