MicrochipTech/cryptoauthlib #260: atcab_ecdh_enc call has not considered ATCA_USE_CONSTANT_HOST_NONCE case

rspworks, 2021-09-20:
Hello Bryan,

https://github.com/MicrochipTech/cryptoauthlib/blob/b177b31182573f439191214d356a34f5db048beb/lib/pkcs11/pkcs11_key.c#L693

This function will give a compilation error if ATCA_USE_CONSTANT_HOST_NONCE is used.

I think this ATCA_USE_CONSTANT_HOST_NONCE macro can be avoided by using host nonce if the nonce parameter is NULL

bryan-hunt, 2021-09-20:
ATCA_USE_CONSTANT_HOST_NONCE is unsafe and only exists for people that are using the library directly in a legacy environment. PKCS11 is technically an application of this library and it is incompatible with this option.

closed 2021-09-20 by bryan-hunt

rspworks, 2021-09-20:
Ok got it. 
Thanks.

rspworks, 2021-09-20:
Sorry to confuse you again. But in this file last parameter is passed as NULL 
There are no NULL checks till it ends up in memcpy in the last function.

Please clarify.