MicrochipTech/cryptoauthlib #263: Use correct #ifdef instead of #if 0 for pkcs11_slot_check_device_type

rspworks, 2021-09-22:
Hello Bryan,

https://github.com/MicrochipTech/cryptoauthlib/blob/b177b31182573f439191214d356a34f5db048beb/lib/pkcs11/pkcs11_slot.c#L122

pkcs11_slot_check_device_type is commented out in 3.3.2 

I think, instead, we need the below check as its caller has the same check 

    #if PKCS11_508_SUPPORT && PKCS11_608_SUPPORT

My application needs support for both 508A, 608A,  608B  simultaneously (ie same build for all device types)

Thanks,
Rohan

bryan-hunt, 2021-09-22:
A function needs to be called by code in order to make an impact in software. This function is not called by any code.

closed 2021-09-22 by bryan-hunt

rspworks, 2021-09-22:
Yes. it is getting called by  pkcs11_slot_init() 

lib/pkcs11/pkcs11_slot.c line 222

bryan-hunt, 2021-09-22:
PKCS11_508_SUPPORT and PKCS11_608_SUPPORT do nothing and are never set by the resulting pkcs11 configuration. The library expects that the device be specified in the slot configuration file. 

https://github.com/MicrochipTech/cryptoauthlib/blob/b177b31182573f439191214d356a34f5db048beb/app/pkcs11/slot.conf.tmpl#L17

rspworks, 2021-09-22:
Ok got it. 

So this means 'device ' needs to be configured statically in .conf file.

In my use case, I have all hardware platforms (508A, 608A, 608B). I am expecting it to be worked for all.

Note: I am using  features based on device type received in atcab_info in info[2]