MicrochipTech/cryptoauthlib #275: Zephyr integration unclear

cafe88, 2021-11-10:
Hi!

I want to use cryptoauthlib together with Zephyr but the instructions for doing this do not work for me.

My CMakeLists.txt:

```cmake
cmake_minimum_required(VERSION 3.20.0)

# Set some specific configurations.
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
set(ZEPHYR_TOOLCHAIN_VARIANT "gnuarmemb")

find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(example)

add_subdirectory(cryptoauthlib/lib)

# Zephyr already added the executable by invoking find_package().
# A Zephyr target is always called 'app'!

# Add sources.
target_sources(app
PRIVATE 
    src/main.c
)

# Add includes here.
target_include_directories(app
PRIVATE
    src
    include
)
```

I have no problem configuring, but when I build the application, I get an error:

```
cryptoauthlib/third_party/hal/zephyr/hal_zephyr.c:9:10: fatal error: zephyr.h: No such file or directory
```

Do you have an example on how to implement Cryptoauthlib into zephyr?

Thanks in advance,
Martin

bryan-hunt, 2021-11-10:
I'm not exactly sure why the paths are incorrect. Here is what I've been using:

```
cmake_minimum_required(VERSION 3.13.1)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
project(zephyr-cryptoauth-example)

# Configure cryptoauthlib
SET(ATCA_BUILD_SHARED_LIBS OFF CACHE INTERNAL "")
SET(ATCA_PRINTF ON CACHE INTERNAL "")
SET(ATCA_HAL_I2C ON CACHE INTERNAL "")
SET(ATCA_NO_HEAP ON CACHE INTERNAL "")
add_subdirectory(cryptoauthlib/lib)

include_directories(app ${CMAKE_CURRENT_BINARY_DIR}/cryptoauthlib/lib 
                        ${CMAKE_CURRENT_SOURCE_DIR}/cryptoauthlib
                        ${CMAKE_CURRENT_SOURCE_DIR}/cryptoauthlib/lib
                        ${CMAKE_CURRENT_SOURCE_DIR}/cryptoauthlib/third_party/hal/zephyr)
                  
target_link_libraries(cryptoauth PUBLIC zephyr_interface)
target_link_libraries(app PUBLIC cryptoauth)
```

cafe88, 2021-11-19:
Hi!

Yes, i am using west for configuration.
Now the code is compiling successfully.
I think it would be nice to make a complete zephyr module out of the cryptoauthlib if this is possible with all needed includes and links made inside the CMakeLists.txt of the library?
Like that, no one needs to worry about what include directories to configure or when to link which library.

But do you have the complete Zephyr example somewhere accessible for me?
Next, there is a gap on how to configure the device properly and not hacky.
Half of the configuration has to be done in Devicetree, the other half is done via library configuration.

It would be nice if you may send me you complete example if this is ok for you so i can check whats wrong on my side by myself (Currently, I get an MPU-Error somewhere in the initialization).

Thank you for your quick reply and hoping to get it all running soon ;-)

closed 2022-11-13 by github-actions[bot]