MicrochipTech/cryptoauthlib #286: Asymmetric Encryption using ATECC608B (TFLEX)

GalaxyGorilla, 2022-02-25:
Hi, sorry for not using the official path here but I think my question is a bit out of scope anyway.

Is it possible to use asymmetric encryption using the primary key (Slot 0) of the the ATECC608B using the TFLEX configuration? Maybe using other key slots?

From the code I can't see any support :-/. I think there's no support at all for this in the device.

The background here is that I maintain an [Erlang wrapper library for cryptoauthlib](https://github.com/grisp/grisp_cryptoauth) and a use case popped up where it might make sense to encode disk encryption keys using asymmetric encryption (server encrypts OTA update keys using public key from the ATECC608B).

Hope asking that question here is fine, might be nice to know for other people too.

Cheers

GalaxyGorilla, 2022-02-25:
After some research I found out that ECC (in opposite to RSA) is not suited for what I'm aiming for here.

Sorry for the fuzz.

closed 2022-02-25 by GalaxyGorilla

bryan-hunt, 2022-02-25:
@GalaxyGorilla correct one should use ECIES if you want to use elliptic curve to transfer encrypted data blobs. However either using ECIES or RSA in this situation opens another issue - how do you verify the sender of the encryption key is not an attacker attempting to trigger your system into doing something irreversible by using a bad encryption key? While not strictly according to the standard if you use fixed private keys in an ECIES scheme you do get mutual authentication if you verify the public key of the sender with a certificate chain first.

GalaxyGorilla, 2022-02-28:
@bryan-hunt thank you for this insight, I was not aware of ECIES and over the weekend I discovered some nice reads about this topic. Is the ATECC capable of supporting ECIES? I can see that there is some support for KDFs but it is not explicitly mentioned anywhere.