MicrochipTech/cryptoauthlib #288: Linux UART does not time out while receiving (Raspberry PI)

StWandl, 2022-03-23:
Hello everybody,

I successfully built and used the library on my Raspberry PI.
But one thing I noticed is that `read()` in `hal_uart_receive()` (_hal_linux_uart_userspace.c_) did not time out when nothing is received. So it just blocked my application.

In hal_uart_open_file() the read timeout is set to 500ms.
`// hal_linux_uart_userspace.c line 146`
`tty.c_cc[VTIME] = 5;`

But read() still does not time out, so I did some digging and found a solution by setting VMIN in the termios struct to zero.
`tty.c_cc[VMIN] = 0;`

Some infos for VTIME and VMIN in the termios struct:
[http://unixwiz.net/techtips/termios-vmin-vtime.html]()

VMIN represents the minimum number of received bytes.
So if VMIN > 0 --> read() never times out because nothing is received.

With VMIN set to 0 the read() call timed out after 500ms.

So I think setting VMIN to 0 should be added to `hal_uart_open_file()`.
Or did I miss something that would be negatively affected by this change?

bryan-hunt, 2022-03-23:
It would have to be tested - there are a couple of usecases for the uart usage. Kit protocol (which is basic ascii frames) and SWI which is more complicated (oversampling). In the later case the rx & tx lines are tied together.

Were you using kit protocol to communicate with a development board (i.e. with the kit_host from the app folder on the other side) or were you trying to work with a SWI device?

StWandl, 2022-03-25:
Sorry, I forgot to mention. I was communicating with a SWI device (ATSHA204A) and had RX & TX tied together.
The communication worked fine, but one time I forgot to power the device and I noticed that the program got stuck due to the blocking read() call.