MicrochipTech/cryptoauthlib #289: Facing the issue in reading info from TPM module.

kiranjenex, 2022-03-31:
Hi bryan-hunt,

Facing the issue and the baud rate is set to 100000Hz. i2c-detect works properly like
sudo i2cdetect -r -y 1
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- 35 36 -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

but when I run the test app with command
./cryptoauth_test info -d ecc608 -i i2c 1 -a 6C

/home/revofi/cryptoauthlib/lib/calib/calib_read.c:103:f0:calib_read_zone - execution failed
/home/revofi/cryptoauthlib/lib/calib/calib_read.c:619:f0:calib_read_zone - falied
atcab_init() failed with ret=0x000000F0

it displays the error message I debug the code and I found that calib_wake returns the value 0xF0. which means i2c device did not wake up but the i2cdetect works which is confusing.

Please help me out on this issue.
Thanks,
Kiran

bryan-hunt, 2022-04-13:
i2cdetect is extremely simple - it is simply probing the i2c bus looking for acks to the address write. It's really not possible to determine what is going on in this situation other than confirming that you enabled ATCA_HAL_I2C with cmake before you built the library.

Beyond that it's a system/hardware issue and you'll have to submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work with you on your specific hardware.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2022-11-13 by github-actions[bot]

majnun1337, 2022-12-16:
I'm facing exactly the same issue on Raspberry Pi.

kiranjenex, 2022-12-19:
Hello majnum1337,
Please use below steps and try with your TPM module

`sudo apt-get install cmake libudev-dev`

```
cd ~
git clone https://github.com/MicrochipTech/cryptoauthlib.git
cd cryptoauthlib
```

`cmake -S .. -B . -DATCA_ATECC608A_SUPPORT=ON -DBUILD_TESTS=ON -DATCA_PRINTF=ON -DATCA_BUILD_SHARED_LIBS=ON -DATCA_HAL_I2C=ON -DATCA_PKCS11=ON -DATCA_USE_ATCAB_FUNCTIONS=ON
`
```
make 
sudo make install
```
Make sure you use trust and go platform and try to run the below command

`./cryptoauth_test rand -d ecc608 -i i2c 1 -a 0x6c`
`./cryptoauth_test sernum -d ecc608 -i i2c 1 -a 0x6c`