MicrochipTech/cryptoauthlib #293: ATECC108A lock command not work

hoangduy1505, 2022-05-26:
Hi, 

I am using a ATECC508A for our printer, I work good. But now I can not buy new ATECC508A so I need to change to ATECC108A. 
I tested with my printer, so when I send a lock command to lock OTP zone it return an error C508_BAD_CRC. But I am using ignore CRC check command on bit 7. ON THIS FIRMWARE AND HARDWARE RUN WELL ON ATECC508A. 
Can you help me check this problem 
This is my code 
uint8_t c508m_lock_zone_config(uint8_t *tx_buffer, uint8_t *rx_buffer, uint8_t zone, uint16_t address) 
{ 
      uint8_t rx_size, ret_code; 
      
      if (!tx_buffer || !rx_buffer || ((zone & ~READ_ZONE_MASK) != 0) 
      || ((zone & READ_ZONE_MODE_32_BYTES) && (zone == C508_ZONE_OTP))) 
      return C508_BAD_PARAM; 
      
      address >>= 2; 
      if ((zone & C508_ZONE_MASK) == C508_ZONE_CONFIG) 
      { 
      if (address > C508_ADDRESS_MASK_CONFIG) 
      return C508_BAD_PARAM; 
      } 
      else if ((zone & C508_ZONE_MASK) == C508_ZONE_OTP) 
      { 
      if (address > C508_ADDRESS_MASK_OTP) 
      return C508_BAD_PARAM; 
      } 
      else if ((zone & C508_ZONE_MASK) == C508_ZONE_DATA) 
      { 
      if (address > C508_ADDRESS_MASK) 
      return C508_BAD_PARAM; 
      } 
      tx_buffer[C508_COUNT_IDX] = 7; //t[0] = 0x07 
      tx_buffer[C508_OPCODE_IDX] = C508_LOCK; //t[1] = 0x17 
      tx_buffer[C508_PARAM1_IDX] = 0x81; // Lock OTP, none check CRC 
      tx_buffer[C508_PARAM2_IDX] = 0x00;//(uint8_t) (address & C508_ADDRESS_MASK); //t[3] = //addr & 7F 
      tx_buffer[C508_PARAM2_IDX + 1] = 0x00; //t4 = 0 
      
      rx_size = LOCK_RSP_SIZE;
      ret_code = c508c_send_and_receive(&tx_buffer[0], rx_size, &rx_buffer[0], LOCK_DELAY, LOCK_EXEC_MAX - LOCK_DELAY); 
      return ret_code; 
}

hoangduy1505, 2022-05-26:
I sent a ticket support to Microchip but dont get any feedback from Microchip
My ticket ID: 00995030

bryan-hunt, 2022-05-26:
Issues opened here are for use of this library. The above code does not come from this library. Also if you have defined C508_BAD_CRC to be the value 0xD4 (212) see https://github.com/MicrochipTech/cryptoauthlib/blob/9a37b8d685c351faed720d18d9610345bdaf20b0/lib/atca_status.h#L52 then that is in indication that the CRC of the packet was bad and does not have anything to do with the internal data of the packet.

 I have checked the support case and I see that you received a response within the specified SLA of your account. As this is a product support case all future communication must be done there.

closed 2022-05-26 by bryan-hunt

hoangduy1505, 2022-05-26:
Thank Bryan, The C508_BAD_CRC is 0xe5. It return fail CRC when lock command bit 7 set to 1, it mean ignore check CRC of all data. This code work well on ATECC508A but not work on ATECC108A

bryan-hunt, 2022-05-26:
That code does not come from a device - it comes from the software you're using the software code you provided is not from this library. This issue is closed and locked. You have an open microchip support ticket - you need to use that.