MicrochipTech/cryptoauthlib #302: M1 Pro 14" doesn't work with cryptoauthlib

johnsmithecon, 2022-08-08:
**Describe the bug**
cryptoauthlib unusable with the 14" m1 pro mac it seems. I've tried to use previous commit versions with no luck. The error I always get is as follows:
Traceback (most recent call last):
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/library.py", line 100, in load_cryptoauthlib
    _CRYPTO_LIB = cdll.LoadLibrary(library_file)
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/ctypes/__init__.py", line 451, in LoadLibrary
    return self._dlltype(name)
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.dylib, 0x0006): tried: '/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/library.py", line 103, in load_cryptoauthlib
    _CRYPTO_LIB = cdll.LoadLibrary(_force_local_library())
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/ctypes/__init__.py", line 451, in LoadLibrary
    return self._dlltype(name)
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.dylib, 0x0006): tried: '/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/__init__.py", line 23, in <module>
    raise error
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/__init__.py", line 20, in <module>
    load_cryptoauthlib()
  File "/Users/philipperivard/miniforge3/envs/py38/lib/python3.8/site-packages/cryptoauthlib/library.py", line 105, in load_cryptoauthlib
    raise LibraryLoadError('Unable to find cryptoauthlib. You may need to reinstall')
cryptoauthlib.exceptions.LibraryLoadError: Unable to find cryptoauthlib. You may need to reinstall

**To Reproduce**
Steps to reproduce the behavior:
1. Use 14" M1 pro Macbook Pro
2. Use Python 3.8 or 3.9
3. Pip install cryptoauthlib
4. Run "python" command in terminal
5. Run "import cryptoauthlib" command in terminal
6. See error

**Expected behavior**
Should be able to import cryptoauthlib in python

fertrevino, 2022-09-18:
Having the same issue, previous workarounds in the repository (using vscode for x86) did not work.

bryan-hunt, 2022-11-05:
This should have been resolved in the most recent versions of the library

closed 2022-11-05 by bryan-hunt