MicrochipTech/cryptoauthlib #303: Compilation Fails using Visual Studio 2017 with -DATCA_HAL_CUSTOM=ON

nimanp, 2022-09-15:
Hello,

I am trying to include cryptoauthlib in a project and I need to define my own HAL. It looks like the correct way to accomplish this is to run cmake like so:

`cmake -DATCA_HAL_CUSTOM=ON ..`

But when I try to compile using Visual Studio 2017, I get a syntax error on [line 180 of atca_hal.c](https://github.com/MicrochipTech/cryptoauthlib/blob/main/lib/hal/atca_hal.c#L180):

`C:\projects\cryptoauthlib\lib\hal\atca_hal.c(180): error C2059: syntax error: '}'`

Seems to me that VStudio doesn't like having an array of unspecified size that's empty, and there isn't an `#ifdef` in that array for ATCA_HAL_CUSTOM.

Running cmake the exact same way on Ubuntu 20.04 works and I can compile without issues. Is there something I'm missing on Windows? Or is this a defect?

Thanks

bryan-hunt, 2022-09-15:
It shouldn't be of unspecified size (see ATCA_MAX_HAL_CACHE). Regardless its not recommended to use ATCA_HAL_CUSTOM. It is better to use one of the defined HAL interface forms even if you write your own HAL for it.

nimanp, 2022-09-15:
Thank you for the quick response!

I'm trying to define my own i2c HAL on Windows. I was hoping to build vanilla cryptoauthlib and then link against the DLL and provide custom HAL functions in my project. Since there isn't a built-in I2C HAL for Windows, when I try to configure cmake with `-DATCA_HAL_I2C=ON`, it fails to generate the DLL because of unresolved externals:

```
2>atca_hal.obj : error LNK2001: unresolved external symbol _hal_i2c_init
2>atca_hal.obj : error LNK2001: unresolved external symbol _hal_i2c_post_init
2>atca_hal.obj : error LNK2001: unresolved external symbol _hal_i2c_send
2>atca_hal.obj : error LNK2001: unresolved external symbol _hal_i2c_receive
2>atca_hal.obj : error LNK2001: unresolved external symbol _hal_i2c_control
2>atca_hal.obj : error LNK2001: unresolved external symbol _hal_i2c_release
```

Is it possible to do what I'm trying to do? Or will I have to pull the cryptoauthlib source into my project and add my custom HAL to it rather than trying to link against the vanilla DLL?

Hopefully what I'm asking makes sense.

bryan-hunt, 2022-09-15:
Well cryptoauthlib is a cmake project to make that an easier process - at the very least you can incorporate your custom hal with cryptoauthlib into the same project to produce a single DLL.

Otherwise patch cryptoauthlib to define those symbols with __declspec(dllimport)

closed 2022-11-22 by github-actions[bot]