MicrochipTech/cryptoauthlib #307: Sending both the device and signer certificate to the backend

maelp, 2022-10-12:
Hi,

we are using ATECC608 chips with ESP32 for our product, and we want to use them to securely communicate to our backend.

During the provisioning process, MicroChip created many "Device Signer" CA which have been signed by our root CA, and which have been used to sign each device certificate.

Now, when we want to do a mutual TLS connection between the device and the backend, we are able to use the device private key and the device certificate using a simple cryptoauthlib call, but this means that on the backend, we need our server to have all the certificates for all the Device Signers, in addition to the root CA, in order to validate the chain.

Is there a way using cryptoauthlib to have the device send both its device certificate and its signer certificate as a bundle, so the server could verify it using only the root CA?

Thanks

bryan-hunt, 2022-10-12:
No there is not - this is a problem with how you are utilizing your TLS stack. Your stack expects certificates to be provided in a specific way for them to get stapled together. Simply concatenating two der blobs won't work. You need to read the device and signer certs separately and then use whatever mechanism your TLS stack expects to parse and attach them to the TLS session context. For example mbedtls creates a linked list of the required certifictes.

closed 2022-10-12 by bryan-hunt

maelp, 2022-10-12:
Would it be possible to have more details? For now we're able to convert those certificates as PEM and to concatenate them together, which is what my TLS stack is expecting (NGINX with a reverse mTLS proxy), but I'm not sure how to do a call that use those certificates while also using the chip in order to access the private key and do the TLS communication properly? Would you have some example code?

maelp, 2022-10-12:
I guess it's a common scenario when using Microchip TrustFlex elements no, so there should perhaps be some code somewhere to demonstrate how to do this?

bryan-hunt, 2022-10-12:
That's really up to the design of your system - without knowing what tls stack you are using on your device I can't provide any guidance other than to read the documentation of the tls stack. Cryptoauthlib provides the tools to read the certificates and every tls stack has their own way of parsing and attaching certificates to their tls context. WolfSSL has native support cryptoauthlib mechanism and cryptoauthlib comes with integrations to help with mbedtls.

For example the general outline for establishing mbedtls is along the form of this:

```
    int ret;
    mbedtls_net_context server_fd;
    const char *pers = "ssl_client1";

    mbedtls_entropy_context entropy;
    mbedtls_ctr_drbg_context ctr_drbg;
    mbedtls_ssl_context ssl;
    mbedtls_ssl_config conf;
    mbedtls_x509_crt cacert;
    mbedtls_pk_context pkey;
    mbedtls_x509_crt cert;
    ATCA_STATUS status;

	 /* Start a session with the device */
	 if (ATCA_SUCCESS != (status = atcab_init(&cfg_ateccx08a_kithid_default)))
	 {
	     printf("Failed to init: %d\r\n", status);
	 }

    /*
    * 0. Initialize the RNG and the session data
    */
    mbedtls_net_init(&server_fd);
    mbedtls_ssl_init(&ssl);
    mbedtls_ssl_config_init(&conf);
    mbedtls_x509_crt_init(&cacert);
    mbedtls_x509_crt_init(&cert);
    mbedtls_ctr_drbg_init(&ctr_drbg);

    printf("\n  . Seeding the random number generator...");
    fflush(stdout);

    mbedtls_entropy_init(&entropy);
    if ((ret = mbedtls_ctr_drbg_seed(&ctr_drbg, mbedtls_entropy_func, &entropy,
        (const unsigned char *)pers,
        strlen(pers))) != 0)
    {
        printf(" failed\n  ! mbedtls_ctr_drbg_seed returned %d\n", ret);
        goto exit;
    }

    printf(" ok\n");

    /* Set Up Defaults */
    printf("  . Setting up the SSL/TLS structure...");
    fflush(stdout);

    if ((ret = mbedtls_ssl_config_defaults(&conf,
        MBEDTLS_SSL_IS_CLIENT,
        MBEDTLS_SSL_TRANSPORT_STREAM,
        MBEDTLS_SSL_PRESET_DEFAULT)) != 0)
    {
        printf(" failed\n  ! mbedtls_ssl_config_defaults returned %d\n\n", ret);
        goto exit;
    }

    printf(" ok\n");

    /* Convert to an mbedtls key */
    if (0 != atca_mbedtls_pk_init(&pkey, 0))
    {
        printf("Failed to parse key from device\n");
        goto exit;
    }

    /* Extract the device certificate and convert to mbedtls cert */
    if (0 != atca_mbedtls_cert_add(&cert, &g_cert_def_2_device))
    {
        printf("Failed to parse cert from device\n");
        goto exit;
    }

    /* Extract the signer certificate, convert, then attach to the chain */
    if (0 != atca_mbedtls_cert_add(&cert, &g_cert_def_1_signer))
    {
        printf("Failed to parse cert from device\n");
        goto exit;
    }

    /* Attach the certificate chain and private key to the SSL/TLS context */
    printf("  . Set up the client credentials.");
    fflush(stdout);
    if(0 != (ret = mbedtls_ssl_conf_own_cert(
            &conf, &cert, &pkey)))
    {
        printf(" failed\n ! mbedtls_ssl_conf_own_cert returned %d\r\n", ret);
        goto exit;
    }
    printf(" ok\n");

    mbedtls_ssl_conf_authmode(&conf, MBEDTLS_SSL_VERIFY_OPTIONAL);
    mbedtls_ssl_conf_rng(&conf, mbedtls_ctr_drbg_random, &ctr_drbg);
    mbedtls_ssl_conf_dbg(&conf, my_debug, stdout);

    /* Set up the "ssl" session */
    if ((ret = mbedtls_ssl_setup(&ssl, &conf)) != 0)
    {
        printf(" failed\n  ! mbedtls_ssl_setup returned %d\n\n", ret);
        goto exit;
    }

    if ((ret = mbedtls_ssl_set_hostname(&ssl, endpoint)) != 0)
    {
        printf(" failed\n  ! mbedtls_ssl_set_hostname returned %d\n\n", ret);
        goto exit;
    }

    /* Start the connection */
    printf("  . Connecting to tcp/%s/%s...", endpoint, port);
    fflush(stdout);

    if ((ret = mbedtls_net_connect(&server_fd, endpoint, port, MBEDTLS_NET_PROTO_TCP)) != 0)
    {
        printf(" failed\n  ! mbedtls_net_connect returned %d\n\n", ret);
        goto exit;
    }

    printf(" ok\n");

    /* Attach the open handle to the ssl context */
    mbedtls_ssl_set_bio(&ssl, &server_fd, mbedtls_net_send, mbedtls_net_recv, NULL);

    /* Start the tls handshake (opens the socket itself) */
    printf("  . Performing the SSL/TLS handshake...");
    fflush(stdout);

    while ((ret = mbedtls_ssl_handshake(&ssl)) != 0)
    {
        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE)
        {
            printf(" failed\n  ! mbedtls_ssl_handshake returned -0x%x\n\n", -ret);
            goto exit;
        }
    }
    printf(" ok\n");

    printf("TLS Session Established and a Socket is ready for an Application\n");
    /* @TODO: Insert application here to interact with the now open TLS socket */

    mbedtls_ssl_close_notify(&ssl);

exit:

    mbedtls_net_free(&server_fd);

    mbedtls_x509_crt_free(&cacert);
    mbedtls_ssl_free(&ssl);
    mbedtls_ssl_config_free(&conf);
    mbedtls_ctr_drbg_free(&ctr_drbg);
    mbedtls_entropy_free(&entropy);

    return ret;
```

There are also examples people have done like this: https://github.com/PBearson/esp-aws-iot/blob/master/port/network_mbedtls_wrapper.c

Regardless that is the extent of the feedback that can be provided here. This is not problem with the library - for product and application support you need to use the [Microchip Support Portal](https://microchipsupport.force.com/s/).