MicrochipTech/cryptoauthlib #308: How to solve blocking calls inside cryptoauthlib?

LumbirBwut, 2022-10-21:
Hi, 
We are currently using the ATECC608 TFLX on our device, which is driven by an ESP32, and where the ATECC shares its bus with another I2C device.

To match our need we currently forked esp-cryptoauthlib, where we shared a mutex between cryptoauthlib and our codebase to protect each calls inside of `hal_esp32_i2c.c`. This allow us to have concurrent calls on our I2C bus and seems to be working fine. 

However we noticed that for some calls inside cryptoauthlib (for eg `calib_wakeup_i2c`) as the function uses a certain numbers of retries (20 as defined in the configuration), it could block our other threads from doing their stuff in the meantime, for up to 20s, which is not acceptable in our situation.

Is there any mechanisms inside cryptoauthlib, such as putting it to sleep between retries, which could allow us to avoid it blocking for extended periods of time, and let other threads do their work ?

bryan-hunt, 2022-10-21:
What you're asking for is not the definition of non-blocking. The call will still be blocking but it will work with the rtos to sleep the thread rather than spining. Include the `lib/hal/hal_freertos.c` file into your project and define the following in `atca_config.h`:

```
#define ATCA_USE_RTOS_TIMER     (1)
#define atca_delay_ms   hal_rtos_delay_ms
#define atca_delay_us   hal_delay_us
```

If you don't have INCLUDE_xTaskGetSchedulerState  defined in your freertos configuration then you will not be able to call cryptoauthlib functions before the scheduler is running.

LumbirBwut, 2022-11-09:
Hi @bryan-hunt thanks for your quick answer, your solution seems to match our needs so far, we'll still need to do some test on our side though.

Side question quite related, inside `calib_wakeup_i2c` function there is a retry mechanism. I observe that in a threaded environnement when other threads are doing more intense work on their sides _(for eg: doing an OTA update)_ the call to functions of the cryptochip will be more likely to fail. What is problematic is that if the functions calls to cryptochip occurs as part of an ssl_handshake  _(which requires multiple successive calls to cryptochip function)_ the overall result will more likely fail.

I considered two solutions, give more time for the cryptochip to do its calculation _(aka put in longer sleep the other threads)_, it seems to have some effect but not really what we want on our side OR modify some function in cryptoauthlib which are subect to failure _(for eg: calib_read_zone)_ to incorporate a similar mechanism as the one found in `calib_wakeup_i2c`.

Here is an example of what it looks:

```c
ATCA_STATUS calib_read_zone(ATCADevice device, uint8_t zone, uint16_t slot, uint8_t block, uint8_t offset, uint8_t *data, uint8_t len)
{
    ATCAPacket packet;
    ATCA_STATUS status = ATCA_GEN_FAIL;
    uint16_t addr; 
    ATCAIface iface = atGetIFace(device);  // ADDED (I should test if it's null or not though)
    uint8_t retries = atca_iface_get_retries(iface); // ADDED

    do
    {
        // Check the input parameters
        if ((device == NULL) || (data == NULL))
        {
            status = ATCA_TRACE(ATCA_BAD_PARAM, "NULL pointer received");
            break;
        }

        if (len != 4 && len != 32)
        {
            status = ATCA_TRACE(ATCA_BAD_PARAM, "Invalid length received");
            break;
        }

        // The get address function checks the remaining variables
        if ((status = calib_get_addr(zone, slot, block, offset, &addr)) != ATCA_SUCCESS)
        {
            ATCA_TRACE(status, "calib_get_addr - failed");
            break;
        }

        // If there are 32 bytes to read, then OR the bit into the mode
        if (len == ATCA_BLOCK_SIZE)
        {
            zone = zone | ATCA_ZONE_READWRITE_32;
        }

        // build a read command
        packet.param1 = zone;
        packet.param2 = addr;

        if ((status = atRead(atcab_get_device_type_ext(device), &packet)) != ATCA_SUCCESS)
        {
            ATCA_TRACE(status, "atRead - failed");
            break;
        }

        if ((status = atca_execute_command(&packet, device)) != ATCA_SUCCESS)
        {
            ATCA_TRACE(status, "calib_read_zone - execution failed");
            // break;  // MODIFIED
        }
        if (status == ATCA_SUCCESS)  // ADDED
            memcpy(data, &packet.data[1], len);
        
        atca_delay_ms(50);  // ADDED
    }
    while (0 < retries-- && ATCA_SUCCESS != status);  // MODIFIED

    return status;
}

```

In our threaded environnement the retry mechanism is not a blocking problem for us, and it seems to increase the overall chance of success of our ssl_handshake, hence my question is a good idea to do so ?

Thanks in advance

bryan-hunt, 2023-01-06:
This is a redundant operation as atca_execute_command is already performing this operation. Only the pkcs11 api has thread/process safety built in - all other APIs are not thread safe. Based on your description you're simply observing concurrent threads interfering with each other. You need to institute guards for your cryptoauthlib calls.

closed 2023-01-06 by bryan-hunt