MicrochipTech/cryptoauthlib #310: pkcs11 related operations fail due to "xmalloc: zero size"

Irdroid, 2022-10-31:
When using the cryptochip ATECC608B , from Microchip, with the provided cryptolibrary "cryptoauthlib", the pkcs11 related operations fail due to "xmalloc: zero size".

Steps to reproduce:
1.- Compile and install Microchip cryptoauthlib library.
2.- *Modify this library to handle unset Mutexes (in functions pkcs11_lock_context and pkcs11_unlock_context change rv = CKR_CANT_LOCK for rv = CKR_OK)*
3.- execute command ssh-keygen -D /usr/lib/libcryptoauth.so

Actual result:
xmalloc: zero size

Expected output:
C_GetAttributeValue failed: 7
failed to fetch key
ecdsa-sha2-nistp256 AAAAE2VjZHNhLXNoYTItbmlzdHAyNTYAAAAIbmlzdHAyNTYAAABBBPmKkZ2M7DeVdwOpCW8XSnLYUbPx5RIk8OF8B0F0OwmRWexpsZONwft41YRI76gxZ/cN7wt4wO765ULvXQhxFCQ= device

bryan-hunt, 2022-10-31:
Step 2 doesn't make any sense and points to a problem with the environment.

Irdroid, 2022-10-31:
> Step 2 doesn't make any sense and points to a problem with the environment.

Step 2 is needed to make the pkcs11 provider e.g the cryptoauthlib to work with openssh ( I can provide later the output if no modification to the library is made) . 

The issue is discussed also here:

https://bugzilla.mindrot.org/show_bug.cgi?id=3364[https://bugzilla.mindrot.org/show_bug.cgi?id=3364](https://bugzilla.mindrot.org/show_bug.cgi?id=3364)

bryan-hunt, 2022-11-05:
Please try again with v3.4.0 which was just released

Irdroid, 2022-11-06:
> Please try again with v3.4.0 which was just released

I will do first thing on Monday, thank you

Irdroid, 2022-11-06:
Bryan, It seems that the xmalloc issue is now gone, but I am using a ATECC608B  and on my system it is at /dev/i2c-11 (I set the i2c bus number to 11 in my /var/lib/cryptoauthlib/0.conf ) and when I try to connect , I am getting this output:

C_GetInfo for provider /usr/lib/libcryptoauth.so failed: 400
C_Finalize for provider /usr/lib/libcryptoauth.so failed: 400

if I make a symbolic link /dev/i2c-1 <> /dev/i2c-11 and set i2c-1 bus number in my /var/lib/cryptoauthlib/0.conf then everything is OK. But I guess that something is not okay with parsing the i2c bus number from the config.  

**As a conclusion -** With version 3.4.0 I am able to authenticate to a remote host using the cryptoauthlib as a pkcs11 provider with openssh (Only one warrning appears in the debug info - C_GetAttributeValue failed: 7 )
[log-3.4.0.txt](https://github.com/MicrochipTech/cryptoauthlib/files/9945889/log-3.4.0.txt)

bryan-hunt, 2022-11-06:
The I2C bus value in the configuration file is expected to be a hex value - so if you want to use bus 11 then specify B in the configuration.

bryan-hunt, 2022-11-06:
As for C_GetAttributeValue failure without details as to what attribute they're trying to retrieve it is hard to say what the issue is. It sounds like based on the previously linked issue however that they may be attempting to use CKA_ID but will fall back to a different mechanism if it fails (since CKA_ID is officially an optional field per the pkcs11 specification). If you want to make the error completely go away and they are using the CKA_ID field then you can specify -DPKCS11_AUTO_ID_ENABLE=ON when you invoke cmake for configuration.

Irdroid, 2022-11-06:
Sorted out in version 3.4.0