MicrochipTech/cryptoauthlib #317: Test application can't access to ATECC608A device via I2C [Embedded Linux]

amolchor25, 2022-11-23:
**Bug**
Test application can't access to ATECC608A device via I2C

However, when we are running sernum attribute, test app returns the next lines:
root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c -a 60

atcab_init() failed with ret=0x000000F0

ATECC608 is correctly connected to platform via I2C. Running i2cdetect -r -y 1 command, we can check that ATECC608 has 0x60 address in i2c bus. Result of i2cdetect command.

root@am335x-evm:# i2cdetect -r -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@am335x-evm:~#

Any idea?

karthik-mc, 2022-11-23:
The most obvious issue would be that the address, as defined by the part datasheet which uses the upper 7 bits of the byte to specify the address. If your i2c probe is showing the device as 0x60 then the address specified here would be 0xC0

amolchor25, 2022-11-23:
**Thanks for reply @karthik-mc !!**

I have tried with 0xC0 but giving the same output

root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c -a C0

atcab_init() failed with ret=0x000000F0

root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c -a c0

atcab_init() failed with ret=0x000000F0

root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c -a 0xc0

atcab_init() failed with ret=0x000000F0

root@am335x-evm:~/HPGP#

ietxezarreta5, 2022-11-23:
Try adding the I2C bus number after the bus flag. In your case, bus number 1:

`root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c 1 -a c0`

amolchor25, 2022-11-23:
Thanks for reply @ietxezarreta5 !!

Tried as you mentioned bus number
root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c 1 -a c0

_./cryptoauth_test: symbol lookup error: ./cryptoauth_test: undefined symbol: atcab_read_serial_number_

I have build the package with following command

cmake -DCMAKE_TOOLCHAIN_FILE="../toolchain.cmake" -DBUILD_TESTS=ON -DATCA_ATECC608A_SUPPORT=ON  -DATCA_HAL_I2C=ON -DATCA_ATECC608A_SUPPORT=ON -DBUILD_TESTS=ON -DATCA_PRINTF=ON -DATCA_BUILD_SHARED_LIBS=ON  -DATCA_PKCS11=ON -DATCA_USE_ATCAB_FUNCTIONS=ON ../

Is there any param remaining to add ??

amolchor25, 2022-11-23:
Thanks for reply to all members !!

Currently I have copied the latest libcryptoauth.so in /usr/lib
then it works

**root@am335x-evm:~/HPGP# ./cryptoauth_test sernum -d ecc608 -i i2c 1 -a c0**

**serial number:
01 23 7E 39 B5 03 D6 35 EE**

closed 2022-11-23 by amolchor25