MicrochipTech/cryptoauthlib #319: cannot load Private Key from engine

amolchor25, 2022-12-06:
Hello team,

I am working on am335x board.
having cross-compiled cryptoauthlib with pkcs11 and openssl in the board.
when I tried to generate csr using openssl command it shows below error.

_**root@am335x-evm:~# openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
/usr/lib/engines-1.1/libpkcs11.so: undefined symbol: C_GetFunctionList
Unable to load module /usr/lib/engines-1.1/libpkcs11.so
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3070087184:error:81065401:libp11:pkcs11_CTX_load:Unable to load PKCS#11 module:p11_load.c:77:
3070087184:error:8006B067:pkcs11 engine:func(107):invalid parameter:eng_back.c:603:
3070087184:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:../openssl-1.1.1j/crypto/engine/eng_pkey.c:77:
unable to load Private Key**_

I have configured below conf file
root@am335x-evm:~# cat /var/lib/cryptoauthlib/0.conf 
_# Reserved Configuration for a device
_# The objects in this file will be created and marked as undeletable
_# These are processed in order. Configuration parameters must be comma
_# delimited and may not contain spaces

interface = i2c,0xC0,1
freeslots = 1,2,3

_# Slot 0 is the primary private key
object = private,device,0

_# Slot 10 is the certificate data for the device's public key
_#object = certificate,device,10

_# Slot 12 is the intermedate/signer certificate data
_#object = certificate,signer,12

_# Slot 15 is a public key
object = public,root,15

let me know is there any missing from my configuration.

ietxezarreta5, 2022-12-06:
Hello,

Have you correctly installed libp11 in your system? It seems as if it can not load the libpkcs11.so module...
Could you also provide the openssl configuration you are using?
Also, are you using last version of cryptoauhtlib?

amolchor25, 2022-12-06:
Yes I have installed the libp11 with below configure
./configure --host=arm-none-linux-gnueabihf CC=arm-none-linux-gnueabihf-gcc OPENSSL_CFLAGS='-I/home/amol/Downloads/new/openssl-1.1.1j/pkg/include' OPENSSL_LIBS='-L/home/amol/Downloads/new/openssl-1.1.1j/pkg/lib -lcrypto' --with-enginesdir=/usr/lib/engines-1.1 

below is the openssl.cnf file in /etc/ssl having configuration

[openssl_init] 
engines=engine_section 

[engine_section] 
pkcs11 = pkcs11_section 

[pkcs11_section] 
engine_id = pkcs11 
dynamic_path = /usr/lib/ssl-1.1/engines-1.1/libpkcs11.so 
MODULE_PATH = opensc-pkcs11.so 
init = 0 

root@am335x-evm:~# openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
opensc-pkcs11.so: cannot open shared object file: No such file or directory
Unable to load module opensc-pkcs11.so
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3069923344:error:81065401:libp11:pkcs11_CTX_load:Unable to load PKCS#11 module:p11_load.c:77:
3069923344:error:8006B067:pkcs11 engine:func(107):invalid parameter:eng_back.c:603:
3069923344:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:../openssl-1.1.1j/crypto/engine/eng_pkey.c:77:
unable to load Private Key
root@am335x-evm:~# ls -l /etc/ssl/opensc-pkcs11.so 
lrwxrwxrwx 1 root root 41 Jul 25 17:01 /etc/ssl/opensc-pkcs11.so -> /usr/lib/ssl-1.1/engines-1.1/libpkcs11.so

ietxezarreta5, 2022-12-06:
I would suggest you trying again changing the `MODULE_PATH` to point to libcryptoauth.so (wherever you have it installed) and trying with key `"pkcs11:token=00ABC;object=device;type=private"`.

In latest cryptoauthlib versions, unless you specify it I think the default token label is set to 00ABC (for 0.conf). 

Hope this works for you!

amolchor25, 2022-12-06:
Hi @ietxezarreta5 
I have set the MODULE_PATH and also used the given key

_**root@am335x-evm:~ openssl req -engine pkcs11 -key "pkcs11:token=00ABC;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
Segmentation fault (core dumped)**_

**[pkcs11_section]                                                  
engine_id = pkcs11                                                       
dynamic_path = /usr/lib/ssl-1.1/engines-1.1/libpkcs11.so               
MODULE_PATH = /usr/lib/libcryptoauth.so                             
init = 0**

ietxezarreta5, 2022-12-06:
Hi,
Are you able to list the available tokens using a PKCS#11 client, as for example `p11tool`? Or do you also get segmentation fault?

amolchor25, 2022-12-06:
Hi @ietxezarreta5 
p11tool is not cross-compiled yet.
Could you please mention the git source code for p11tool so I will cross compile for my device.

ietxezarreta5, 2022-12-06:
Hi,

I believe p11tool is part of gnuTLS package, so may be you could find it somewhere here https://www.gnutls.org/download.html, although any PKCS#11 client could be valid i.e. `pkcs11-tool` from OpenSC.

amolchor25, 2022-12-06:
Hi @ietxezarreta5 
I have tried the p11tool
below is the snapshot

[image]

amolchor25, 2022-12-06:
[image]

Hi @ietxezarreta5 
Tried the command which you shared.
 
the cryptoauthlib used is Release/v3.2.x git branch

amolchor25, 2022-12-06:
Hi @ietxezarreta5 

It works with bus number 

[image]

ietxezarreta5, 2022-12-06:
Hi,

And the CSR generation through OpenSSL still fails? If it does, I believe the problem should be in your pkcs11.so library.

Try also if the process fails when using other p11tool commands like 
`p11tool --provider=/usr/lib/libcryptoauth.so --export-pubkey "pkcs11:token=00ABC;object=device;type=private"`
If these are good, I would focus on your libp11 compilation.

Cheers!

bryan-hunt, 2022-12-06:
For product support please submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2022-12-06 by bryan-hunt

amolchor25, 2022-12-07:
[image]
Hi
getting error in pubkey

bryan-hunt, 2022-12-07:
Your device isn't configured. 

For product support please submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)