MicrochipTech/cryptoauthlib #328: ATECC608 and Greengrass V2 integration with PKCS11

majh, 2023-01-20:
I was wondering if anyone has got an ATECC608 device integrated with Greengrass V2, so that Greengrass V2 was able to use the ATECC608 device via PKCS11.
I have seen some commits in cryptoauthlib to support this, and I know that a recent version of greengrass should be required for EC keys - but i'm running v2.9.2, with its PKCS11Provider.jar, and I haven't been able to successfully get this to work.
 Are there any examples of this working ?
I can happily use it with openssl and p11tool.
`
root@tegra:~/gg# p11tool --provider /usr/lib/libcryptoauth.so --list-tokens
Token 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=XXXXXXXXXXX;token=MCHP
        Label: MCHP
        Type: Hardware token
        Manufacturer: Microchip Technology Inc
        Model: ATECC608A
        Serial: XXXXXXXXXXX
        Module: (null)
`

JamieHunter, 2023-01-31:
@majh Sorry for late responding to this. Are you still stuck? Can you give me more detail of symptoms you are seeing / errors in the error log?

majh, 2023-01-31:
Hi  @JamieHunter, thanks for your response. Here are the details I have
* I use v3.4.1 of cryptoauthtools, build and install on raspberry pi 4, raspian 11
* ATECC608A TNG connected to PI via I2C bus 1, header pins 3 (SDA1),  4(SCL1), 3v3/GND. Note that I have also tried on the ATECC608B on a different piece of hardware with the same issues occuring.
* I can query with cryptoauth_test to verify version, and 'tng' tests complete successfully. Note that I need to change baud rate from 400000 to 100000 in [atca_test_config.c](https://github.com/MicrochipTech/cryptoauthlib/blob/main/test/atca_test_config.c#L84)
```
./cryptoauth_test 
$ ecc608
Device Selected.
$ info -i i2c 1 -a 6a
revision:
00 00 60 02

$ tng

Unity test run 1 of 1
TEST(tng_atca, tng_get_device_pubkey) PASS
TEST(tng_atcacert_client, tng_atcacert_root_public_key) PASS
TEST(tng_atcacert_client, tng_atcacert_root_cert) PASS
TEST(tng_atcacert_client, tng_atcacert_max_signer_cert_size) PASS
TEST(tng_atcacert_client, tng_atcacert_read_signer_cert) PASS
TEST(tng_atcacert_client, tng_atcacert_signer_public_key_no_cert) PASS
TEST(tng_atcacert_client, tng_atcacert_signer_public_key_cert) PASS
TEST(tng_atcacert_client, tng_atcacert_max_device_cert_size) PASS
TEST(tng_atcacert_client, tng_atcacert_read_device_cert_no_signer) PASS
TEST(tng_atcacert_client, tng_atcacert_read_device_cert_signer) PASS
TEST(tng_atcacert_client, tng_atcacert_device_public_key_no_cert) PASS
TEST(tng_atcacert_client, tng_atcacert_device_public_key_cert) PASS

-----------------------
12 Tests 0 Failures 0 Ignored 
OK
```

* I can  successfully query details with pkcs11 tools
```
root@raspberrypi# pkcs11-tool --module  /usr/lib/arm-linux-gnueabihf/libcryptoauth.so --list-slots 2> /dev/null
Available slots:
Slot 0 (0x0): 0_3_0
  token label        : device
  token manufacturer : Microchip Technology Inc
  token model        : ATECC608A
  token flags        : rng, token initialized
  hardware version   : 0.2
  firmware version   : 255.255
  serial num         : 23BC526C69CFA001
  pin min/max        : 0/128

root@raspberrypi# p11tool --provider /usr/lib/arm-linux-gnueabihf/libcryptoauth.so --export "pkcs11:object=device;type=cert"  2> /dev/null
-----BEGIN CERTIFICATE-----
MIICIDCCAcWgAwIBAgIQeBNfnqNpyCi/D8yCTlxspDAKBggqhkjOPQQDAjBPMSEw
HwYDVQQKDBhNaWNyb2NoaXAgVGVjaG5vbG9neSBJbmMxKjAoBgNVBAMMIUNyeXB0
byBBdXRoZW50aWNhdGlvbiBTaWduZXIgMjczMDAgFw0yMTA3MjkxNzAwMDBaGA8y
MDQ5MDcyOTE3MDAwMFowQjEhMB8GA1UECgwYTWljcm9jaGlwIFRlY2hub2xvZ3kg
SW5jMR0wGwYDVQQDDBRzbjAxMjNCQzUyNkM2OUNGQTAwMTBZMBMGByqGSM49AgEG
CCqGSM49AwEHA0IABGkvQnJqsCIDzWfsz4u8+rFv4jPIDp7j/GxAOxTVcjoXdnab
8NxShqO0mPJ1TAcydNNvDoUtJ+20vcCsD7amdwujgY0wgYowKgYDVR0RBCMwIaQf
MB0xGzAZBgNVBAUTEmV1aTQ4XzY4MjcxOTRDNDBDRDAMBgNVHRMBAf8EAjAAMA4G
A1UdDwEB/wQEAwIDiDAdBgNVHQ4EFgQUjygzc1g+FTACQivoB+ReVGsatygwHwYD
VR0jBBgwFoAU2Ey4uB9siZ9d68oHMrkkOTW+ZOwwCgYIKoZIzj0EAwIDSQAwRgIh
APCymup1kd0u0FD3z2xOI2xKZT+tGSL1wzAhSjA4wDq4AiEAswc6crO7woapHp+8
6hulGWAb1Lk2KkccuuD9zIHFQqU=
-----END CERTIFICATE-----

root@raspberrypi# pkcs11-tool --module pkcs11-tool --module /usr/lib/arm-linux-gnueabihf/libcryptoauth.so --list-objects 2> /dev/null
Private Key Object; EC
  label:      device
  ID:         8f283373583e153002422be807e45e546b1ab728
  Usage:      sign, derive
  Access:     sensitive, always sensitive, never extractable
  Allowed mechanisms: ECDSA-KEY-PAIR-GEN,ECDSA,ECDSA-SHA256
[... 93 lines cut ...]
      at com.aws.greengrass.deployment.IotJobsClientWrapper.SubscribeToDescribeJobExecutionAccepted(IotJobsClientWrapper.java:198)
      at software.amazon.awssdk.iot.iotjobs.IotJobsClient.SubscribeToDescribeJobExecutionAccepted(IotJobsClient.java:601)
      at com.aws.greengrass.deployment.IotJobsHelper.subscribeToGetNextJobDescription(IotJobsHelper.java:522)
      at com.aws.greengrass.deployment.IotJobsHelper.subscribeToJobsTopics(IotJobsHelper.java:481)
      at com.aws.greengrass.deployment.IotJobsHelper.lambda$setupCommWithIotJobs$5(IotJobsHelper.java:343)
      at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
      at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
      at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
      at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
      at java.base/java.lang.Thread.run(Thread.java:829)
Caused by: software.amazon.awssdk.crt.mqtt.MqttException: Error during getting mqtt connection builder
      at com.aws.greengrass.mqttclient.MqttClient.lambda$new$0(MqttClient.java:184)
      at com.aws.greengrass.mqttclient.MqttClient.lambda$getNewMqttClient$17(MqttClient.java:800)
      at com.aws.greengrass.mqttclient.AwsIotMqttClient.establishConnection(AwsIotMqttClient.java:261)
      ... 14 more
 ```

* I've attached a log of stdout when running greengrass with _-DPKCS11_DEBUG_ENABLE=ON_, where GG appears to be attempting to use the library, but without success.
* I've also attached a log of stdout when used successfully with other pkcs11 tools
* Is there any other info I can provide ?

Thanks for your assistance.

[pkcs11-tool.read-cert.good.txt](https://github.com/MicrochipTech/cryptoauthlib/files/10551384/pkcs11-tool.read-cert.good.txt)
[gg-loader.bad.txt](https://github.com/MicrochipTech/cryptoauthlib/files/10551385/gg-loader.bad.txt)

JamieHunter, 2023-01-31:
Look for an access grant to a debug tool I wrote - https://github.com/JamieHunter/pkcs11-java-test - dependencies will need updating to latest. To enable aws-c-io debugging, set up environment like:
export JAVA_OPTS="-[Djavax.net](http://djavax.net/).debug=all -Daws.crt.log.destination=File -Daws.crt.log.filename=awscrt.log -Daws.crt.log.level=Trace -agentlib:jdwp=transport=dt_socket,server=y,suspend=n,address=*:5005"

This example also enables Java debugging. the file "awscrt.log" will get created in same directory as tool (or same directory as Nucleus).

majh, 2023-02-06:
Confirmed working for me with Greengrass 2.9.2 and ATECC608A/B  with [d4f3b1abe58](https://github.com/MicrochipTech/cryptoauthlib/commit/d4f3b1abe58370c27d53bd8de3558c57f30fbd36) version of cryptoauthlib. It does not work for me with v3.4.1 of cryptoauthlib.
I notice that there are a number of changes in PKCS11 related areas after the working version - these seem to prevent successful operation somehow. I'm not clear on exactly why v3.4.1 works with other pkcs11 tools, ie p11tool/pkcs11-tool, but not Greengrass.

closed 2023-02-06 by majh

JamieHunter, 2023-02-06:
The problem will be related to how different tools interact with the library. In particular, in this case, the pattern of calls and expectations of Java interaction with the library are different to pkcs11-tool's interaction with the library.

@bryan-hunt what will be useful to help debug this?

user-redans, 2023-03-07:
Hi @JamieHunter,

Can you provide us your '/var/lib/cryptoauthlib/0.conf' file?