MicrochipTech/cryptoauthlib #331: Raspberry Pi CM4+ATECC608B issue with p11tool

user-redans, 2023-02-16:
Hi,
For my project I used RaspberryPi CM4 with ATECC608B chip connected on I2C. I tried the latest cryptoauthlib library (v3.4.1) and I have configured:
- ./lib/atca_cfgs.c -> bus 10 and set default baudrate (.atcai2c.baud  = 100000) 
- ./test/atca_test_config.c -> in ATCA_ENABLE_DEPRECATED statement -> bus 10 and baud 100000
After that I have used cmake to create the make file with this flags:

`cmake -S cryptoauthlib -B cryptoauthlib/test/test_build -D ATCA_ATECC608A_SUPPORT=ON -D BUILD_TESTS=ON -D ATCA_PRINTF=ON -D ATCA_BUILD_SHARED_LIBS=ON -D ATCA_HAL_I2C=ON -D ATCA_PKCS11=ON -D ATCA_USE_ATCAB_FUNCTIONS=ON -D ATCA_OPENSSL=ON -D ATCA_TNGTLS_SUPPORT=ON -D ATCA_TNGLORA_SUPPORT=ON -D ATCA_TFLEX_SUPPORT=ON -D ATCA_TNG_LEGACY_SUPPORT=ON`

Next steps I installed and run it with command: ./cryptoauth_test sernum -d ecc608 -i i2c 10 -a 0xC0 -> it gave good serial number from chip. So far it has worked well.

I am trying now to create a token and certificates with p11tool and for that I was inspired from this [tutorial](https://kickstartembedded.com/2022/04/16/raspberry-pi-atecc608-part-2-about-pkcs11-and-testing-mutual-tls-authentication/) 
I understood that 0.conf file must be created in this location /var/lib/cryptoauthlib/ ... so I created it with:

label = MCHP
interface = i2c,0xC0,10
freeslots = 1,2,3
device = ATECC608-TFLXTLS

#object = private,device,0

#object = certificate,device,10

#object = certificate,signer,12

#object = public,root,15

**The results of what I tried:**

pi@raspberrypi:~/cryptoauthlib/test $ p11tool --provider=/usr/lib/arm-linux-gnueabihf/libcryptoauth.so --list-all -d 999
Setting log level to 999
|<2>| p11: Initializing module: /usr/lib/arm-linux-gnueabihf/libcryptoauth.so
./calib/calib_read.c:98:f0:calib_read_zone - execution failed
./calib/calib_read.c:355:f0:calib_read_zone - falied
|<3>| ASSERT: ../../lib/pkcs11.c[scan_slots]:225
|<3>| ASSERT: ../../lib/pkcs11.c[_pkcs11_traverse_tokens]:1577
|<3>| ASSERT: ../../lib/pkcs11.c[_pkcs11_traverse_tokens]:1654
|<3>| ASSERT: ../../lib/pkcs11.c[_gnutls_pkcs11_token_get_url]:2449
warning: no token URL was provided for this operation; the available tokens are:

|<3>| ASSERT: ../../lib/pkcs11.c[scan_slots]:225
|<3>| ASSERT: ../../lib/pkcs11.c[_pkcs11_traverse_tokens]:1577
|<3>| ASSERT: ../../lib/pkcs11.c[_pkcs11_traverse_tokens]:1654
|<3>| ASSERT: ../../lib/pkcs11.c[_gnutls_pkcs11_token_get_url]:2449

pi@raspberrypi:~/cryptoauthlib/test $ p11tool --list-tokens
Token 0:
        URL: pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
        Label: System Trust
        Type: Trust module
        Flags: uPIN uninitialized
        Manufacturer: PKCS#11 Kit
        Model: p11-kit-trust
        Serial: 1
        Module: p11-kit-trust.so

Some suggestions? 
Thanks

bryan-hunt, 2023-02-16:
Bus 10 would be: interface = i2c,0xC0,0x0A

user-redans, 2023-02-17:
I did what you suggested and it worked. Thank you.
But, now I have another issue with initialize:

pi@raspberrypi:~/cryptoauthlib/test/test_build $ p11tool --initialize "pkcs11:serial=23E4643E44467EEE" --label test --provider /usr/lib/arm-linux-gnueabihf/libcryptoauth.so
Enter Security Officer's PIN:
Initializing token...
Error in pkcs11_init:1455: PKCS #11 unsupported feature

**This is what I get in list-all:**

pi@raspberrypi:~/cryptoauthlib/test/test_build $ p11tool --list-all --provider=/usr/lib/arm-linux-gnueabihf/libcryptoauth.so
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=23E4643E44467EEE;token=MCHP;object=device;type=private
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_genkey.c:84:f4:calib_genkey_base - execution failed
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
        Type: Private key (EC/ECDSA)
        Label: device
        Flags: CKA_PRIVATE; CKA_NEVER_EXTRACTABLE; CKA_SENSITIVE;
        ID:

Object 1:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=23E4643E44467EEE;token=MCHP;object=device;type=cert
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
        Type: X.509 Certificate
        Label: device
        ID:

Object 2:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=23E4643E44467EEE;token=MCHP;object=signer;type=cert
./calib/calib_read.c:98:f4:calib_read_zone - execution failed
        Type: X.509 Certificate
        Label: signer
        Flags: CKA_CERTIFICATE_CATEGORY=CA; CKA_TRUSTED;
        ID:

I don't now what trust chip I use, I think is TrustFLEX (TFLXTLS) because it shows:

pi@raspberrypi:~/cryptoauthlib/test/test_build $ pi@raspberrypi:~/cryptoauthlib/test/test_build $ ./cryptoauth_test lockstat -d ecc608 -i i2c 10 -a 0xC0

Config Zone: unlocked
Data Zone  : unlocked

I'am wrong? That's my [chip](https://ro.mouser.com/ProductDetail/Microchip-Technology-Atmel/ATECC608B-SSHDA-B?qs=W%2FMpXkg%252BdQ5IvpsOYxIilQ%3D%3D).
What's the problem with the initialization?

closed 2023-04-25 by github-actions[bot]