MicrochipTech/cryptoauthlib #332: I2C Configuration Status code 225

lissilla, 2023-05-04:
I am having an issue with configuring I2C using the CryptoAuthLib library version 3.3.2 . Specifically, I am experiencing problems with initializing the library and the status variable is returning a code of 225. I have tried configuring the I2C interface using the ATCAIfaceCfg structure and setting the necessary parameters for my GPIO-based interface, but I am still unable to initialize the library correctly.

Can you please provide guidance on how to properly configure the CryptoAuthLib library for use with a GPIO-based I2C interface? 

This is the snippet of code that i am using to configure the library:

`#include <stdint.h>
#include "cryptoauthlib.h"

int main(void) {
    ATCA_STATUS status;
    ATCADevice c;
    uint8_t data[] = {0x01, 0x23, 0x45, 0x67};
    uint8_t read_data[sizeof(data)];
    for (uint8_t i=0; i<0xff; i++){
        ATCAIfaceCfg cfg;
        cfg.iface_type = ATCA_I2C_GPIO_IFACE;
        cfg.devtype = ATECC508A;
        cfg.atcai2c.bus = 0;
        cfg.atcai2c.baud = 57600;
        cfg.atcai2c.address = i;
        cfg.rx_retries = 20;
        status = atcab_init(&cfg);
		printf("status = %d\n", status);
        if (status == ATCA_SUCCESS) {
            printf("Success %d\n", i);
            return 0;
        }
    }
    printf("Nada\n");
    return -1;

}
`

bryan-hunt, 2023-05-04:
This is up to you to provide the interface for you specific GPIO.

See the lib/hal directory for more details on writing your own hal.

SalmanMalik7183, 2023-05-09:
@lissilla is this issue closed?
I am having a same kind of issue integrating atecc608b in arduino framework.

closed 2023-07-16 by github-actions[bot]