MicrochipTech/cryptoauthlib #333: Mutex for I2C is not reserved before communication

AR12345678910, 2023-05-05:
If I do not define `ATCA_HAL_LEGACY_API` the following behavior can be observed:

In `calib_wakeup()` the function `hal_i2c_control()` is called with the parameter `ATCA_HAL_CHANGE_BAUD`.
In calib_idle() the function `hal_i2c_send()` is called.

This is done without selecting a mutex for the I2C via the function `hal_i2c_control()` `ATCA_HAL_CONTROL_SELECT `before.

bryan-hunt, 2023-05-05:
There is no mutex in any of the HALs. What is your platform and expected behavior?

AR12345678910, 2023-05-08:
Hello @bryan-hunt,
I use a freertos based OS.
There are two devices on the I2C bus that should communicate with the controller.

I found the following functions in the documentation (cryptoauthlib-manual.pdf) (`hal_create_mutex, hal_destroy_mutex, hal_lock_mutex, hal_unlock_mutex`).
However, these are not called.
Therefore I assumed in the function (`hal_i2c_control(ATCAIface iface, uint8_t option, void* param, size_t paramlen)`) for the option `ATCA_HAL_CONTROL_SELECT` and `ATCA_HAL_CONTROL_DESELECT` that here a protection of the I2C should be done by a mutex.
Maybe I have interpreted something wrong here?

bryan-hunt, 2023-05-08:
How would the rest of the software know to call those functions?

There is only one API that protects for concurrent access to the cryptoauth device - PKCS11. You can see how pkcs11 is used in the AWS FreeRTOS implementations. There is nothing the library can do to ensure exclusive access to the I2C bus - that is up to the system designer to ensure.

closed 2023-05-08 by bryan-hunt

AR12345678910, 2023-05-08:
Thank you for the quick reply.
The exclusive access to the i2c has now been implemented elsewhere.

maridcx, 2024-01-19:
> Thank you for the quick reply. The exclusive access to the i2c has now been implemented elsewhere.

Hi @AR12345678910. I would like to send you an email. I think you can resolve some questions that I have about this topic. I look forward to your reply. Thanks in advance!