MicrochipTech/cryptoauthlib #336: Feature Request: Espressif Managed Component for wolfSSL

gojimmypi, 2023-06-07:
**Describe the solution you'd like**

I'd like to have this cryptoautlib, more specifically the ECC608 support, available as an Espressif ESP32 [Managed Component in the ESP Registry](https://components.espressif.com/components/espressif/esp-cryptoauthlib/) to be used with [wolfSSL](https://www.wolfssl.com/).

**Describe alternatives you've considered**

This library could possibly be installed as a local ESP-IDF project component, in the `[project]/components/cryptoautlib` directory.

**Additional context**

I'd like to be able to use this from my recently-published [wolfSSL Managed Component](https://components.espressif.com/components/wolfssl/wolfssl).

At the moment, if I add this component to my [wolfSSL project](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_test):

```
idf.py add-dependency "espressif/esp-cryptoauthlib^3.5.1~1"
```

and attempt to compile, I get this error 

```
[..snip]...
-c /mnt/c/workspace/wolfssl-gojimmypi/IDE/Espressif/ESP-IDF/examples/wolfssl_test/managed_components/espressif__espcryptoauthlib/cryptoauthlib/lib/mbedtls/atca_bedtls_wrap.c
/mnt/c/workspace/wolfssl-gojimmypi/IDE/Espressif/ESPIDF/examples/wolfssl_test/managed_components/espressif__esp-cryptoauthlib/cryptoauthlib/lib/mbedtls/atca_mbedtls_wrap.c:882:10: fatal error: mbedtls/pk_internal.h: No such file or directory
  882 | #include "mbedtls/pk_internal.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I see there's mention of the [esp_cryptoauth_utility](https://github.com/espressif/esp-cryptoauthlib/blob/master/esp_cryptoauth_utility/README.md), but there's nothing regarding selection of the underlying cryptography library.

Perhaps there's something already in the works for wolfSSL compatibility? 

If there's interest, I'm certainly willing to help.

bryan-hunt, 2023-06-07:
You need to open this request at the [espressif/esp-cryptoauthlib](https://github.com/espressif/esp-cryptoauthlib) repo

closed 2023-06-07 by bryan-hunt

gojimmypi, 2023-06-07:
Hi @bryan-hunt & thanks for your prompt reply.

Curious. That's an Espressif "release only" fork, with no ability to open issues.

Are there plans for Microchip to have their own managed component for the ESP-IDF?

For comparison: There's been an unofficial [esp-wolfssl](https://github.com/espressif/esp-wolfssl), but it has not been maintained on a timely basis.

What's your level of interest in getting `cryptoauthlib` to work with wolfSSL? It looks like it [did at one time](https://github.com/wolfSSL/wolfssl/blob/c040b4a339f43ccc20469054bb31124c24a57c35/IDE/Espressif/ESP-IDF/examples/wolfssl_test/main/main.c#L65) but there's no [cryptoauthlib.h](https://github.com/wolfSSL/wolfssl/blob/c040b4a339f43ccc20469054bb31124c24a57c35/wolfssl/wolfcrypt/port/atmel/atmel.h#L33) at wolfSSL.

bryan-hunt, 2023-06-07:
@gojimmypi There is good integration support for wolfssl. For more information see [WolfSSL's Interface](https://github.com/wolfSSL/wolfssl/tree/master/wolfcrypt/src/port/atmel). I think its mostly an issue of configuration setups for esp-wolfssl and wolfssl itself. Cryptoauthlib also has a wrapper layer to interface with the wolfssl crypto layers: [wolfssl interface](https://github.com/MicrochipTech/cryptoauthlib/blob/main/lib/wolfssl/atca_wolfssl_interface.c). 

You can see in the esp-cryptoauthlib module that it's enabling all of the mbedtls features of cryptoauthlib rather than the wolfssl ones.