MicrochipTech/cryptoauthlib #341: ATECC608A on Sama7g54 -- Error in pkcs11_init:1455: PKCS #11 unsupported feature

ndivecha, 2023-07-27:
Hello, 

This is my first time dealing with ATECC608A on Sama7g54, so please forgive me if I'm not in right place. I got Sama7g54 EK running with custom Linux OS (build with Yocto). Got all the cryptoauthlib package and I believe I can communicate with ATECC608A. 

```
root@sama7g54:~# p11tool --list-tokens
Token 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=123456789ABCDEFF;token=00ABC
        Label: 00ABC
        Type: Hardware token, Trust module
        Flags: RNG, Uninitialized, uPIN uninitialized
        Manufacturer: Microchip Technology Inc
        Model: ATECC608A
        Serial: 123456789ABCDEFF
        Module: /usr/lib/libcryptoauth.so

root@sama7g54:~# p11tool --list-all
Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=123456789ABCDEFF;token=00ABC;object=device;type=private
        Type: Private key (EC/ECDSA)
        Label: device
        Flags: CKA_PRIVATE; CKA_NEVER_EXTRACTABLE; CKA_SENSITIVE; 
        ID: 

Object 1:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=123456789ABCDEFF;token=00ABC;object=device;type=public
        Type: Public key
        Label: device
        ID: 

Object 2:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=123456789ABCDEFF;token=00ABC;object=device;type=cert
        Type: X.509 Certificate
        Label: device
        ID: 

Object 3:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=123456789ABCDEFF;token=00ABC;object=signer;type=cert
        Type: X.509 Certificate
        Label: signer
        Flags: CKA_CERTIFICATE_CATEGORY=CA; CKA_TRUSTED; 
        ID: 

```
Issue is every time I tried to initialize the chip I get following error `Error in pkcs11_init:1455: PKCS #11 unsupported feature`: 

```
root@sama7g54:~# p11tool --list-token-url
pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=123456789ABCDEFF;token=00ABC
root@sama7g54:~# URL=$(p11tool --list-token-url)
root@sama7g54:~# p11tool --initialize $URL --label=demo -d999                                                                                         
Setting log level to 999
|<2>| p11: Initializing module: p11-kit-trust
|<2>| p11: Initializing module: cryptoauthlib
|<3>| ASSERT: ../../gnutls-3.7.4/lib/pkcs11.c[compat_load]:896
Enter Security Officer's PIN: 1234
Initializing token... |<3>| ASSERT: ../../gnutls-3.7.4/lib/pkcs11_write.c[gnutls_pkcs11_token_init]:1246
|<2>| p11: This operation is not supported

Error in pkcs11_init:1455: PKCS #11 unsupported feature

```

My `0.conf`:
```
root@sama7g54:~# cat /var/lib/cryptoauthlib/0.conf 
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

# Set a label for this slot (optional) - will default to <slot>ABC so
# 0.conf will have a default label 00ABC
#label = MCHP

# Configure the device interface for an enabled HAL
# hid,i2c,<address>
# i2c,<address>,<bus>
# spi,<select_line>,<baud>
interface = i2c,0xC0,1

# Configure the device type - base part number (optional)
device = ATECC608A-TFLXTLS

#Configure open slots for additional pkcs11 objects (optional)
#freeslots = 1,2,3

# Manually configure keys into device locations (slots/handles)

# Slot 0 is the primary private key
#object = private,device,0

# Slot 15 is a public key
#object = public,root,15

```

On Sama7 EK, ATECC is on bus 1, address (0x60). Shifting bit to left by 1 (0xC0). 
```
root@sama7g54:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- UU UU -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --        
```

Any insights will be highly appreciated. 

Cheers, 
Nitesh

bryan-hunt, 2023-07-27:
For product support please submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

In specific your issues are:
- Your part is blank and unconfigured. It is not a ATECC608A-TFLXTLS device. It is best to start with the ATECC608A-TFLXTLS device instead.
- Attempting to configure a part via a tool like p11tool will produce an insecure device which by default is disabled. If you wish to continue with that path you'll have to enable it via configuration option in your yocto recipe.

The above issues will have to be addressed via the listed Microchip support system rather than this repo.

closed 2023-07-27 by bryan-hunt

ndivecha, 2023-07-27:
@bryan-hunt 

Much appreciate for your prompt response. 

Correct, `ATECC608A-TFLXTLS` was just populated automatically and I really did not pay attention to it. I would like to keep the device secure, so are there other tools I have to use to initialize the chip? Just trying to understand the process and do it the right way? I will appreciate if you can direct me to the right channel. 

Cheers,
Nitesh

bryan-hunt, 2023-07-27:
For product support please submit a request through the [mySupport portal](https://microchipsupport.force.com/s/) where Microchip's global team of FAEs can work on your use case and questions.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal]

ndivecha, 2023-07-27:
@bryan-hunt 

My eval board came with `Model: ATECC608A` chip. So I believe I have to provision it myself or use this Trust Platform? I will contact support and see what path they can guide me. 

Much appreciate for your prompt response.