MicrochipTech/cryptoauthlib #342: Compilation issue when BUILD_TESTS in ON.

fittila, 2023-07-30:
Hello!

When cryptoauth_test is building ( option BUILD_TESTS is ON, for example 
cmake -DATCA_HAL_I2C=ON -DATCA_PRINTF=ON -DBUILD_TESTS=ON -S/storage/postal/fittila/cryptoauthlib -B. )
the following errors occur

> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_root_cert_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:81: undefined reference to `tng_atcacert_root_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:85: undefined reference to `tng_atcacert_root_cert'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_root_cert_small_buf_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:95: undefined reference to `tng_atcacert_root_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:100: undefined reference to `tng_atcacert_root_cert'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_read_signer_cert_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:123: undefined reference to `tng_atcacert_max_signer_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:127: undefined reference to `tng_atcacert_read_signer_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:136: undefined reference to `g_tngtls_cert_def_1_signer'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:130: undefined reference to `tng_atcacert_root_public_key'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:136: undefined reference to `g_tngtls_cert_def_1_signer'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_signer_public_key_no_cert_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:178: undefined reference to `tng_atcacert_max_signer_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:182: undefined reference to `tng_atcacert_read_signer_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:187: undefined reference to `g_tngtls_cert_def_1_signer'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:187: undefined reference to `g_tngtls_cert_def_1_signer'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:194: undefined reference to `tng_atcacert_signer_public_key'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_signer_public_key_cert_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:207: undefined reference to `tng_atcacert_max_signer_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:211: undefined reference to `tng_atcacert_read_signer_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:216: undefined reference to `g_tngtls_cert_def_1_signer'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:216: undefined reference to `g_tngtls_cert_def_1_signer'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:223: undefined reference to `tng_atcacert_signer_public_key'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_read_device_cert_no_signer_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:248: undefined reference to `tng_atcacert_max_device_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:252: undefined reference to `tng_atcacert_read_device_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:255: undefined reference to `tng_atcacert_signer_public_key'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:258: undefined reference to `tng_get_device_cert_def'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_read_device_cert_signer_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:307: undefined reference to `tng_atcacert_max_signer_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:311: undefined reference to `tng_atcacert_read_signer_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:314: undefined reference to `tng_atcacert_max_device_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:318: undefined reference to `tng_atcacert_read_device_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:321: undefined reference to `tng_atcacert_signer_public_key'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:324: undefined reference to `tng_get_device_cert_def'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_device_public_key_no_cert_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:370: undefined reference to `tng_atcacert_max_device_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:373: undefined reference to `tng_atcacert_read_device_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:376: undefined reference to `tng_get_device_cert_def'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:386: undefined reference to `tng_atcacert_device_public_key'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_device_public_key_cert_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:400: undefined reference to `tng_atcacert_max_device_cert_size'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:403: undefined reference to `tng_atcacert_read_device_cert'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:406: undefined reference to `tng_get_device_cert_def'
> /usr/bin/ld: /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:416: undefined reference to `tng_atcacert_device_public_key'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_root_public_key_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:59: undefined reference to `tng_atcacert_root_public_key'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_max_signer_cert_size_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:109: undefined reference to `tng_atcacert_max_signer_cert_size'
> /usr/bin/ld: CMakeFiles/cryptoauth_test.dir/tng/tng_atcacert_client_test.c.o: in function `TEST_tng_atcacert_client_tng_atcacert_max_device_cert_size_':
> /storage/postal/fittila/cryptoauthlib/build/test/./test/tng/tng_atcacert_client_test.c:233: undefined reference to `tng_atcacert_max_device_cert_size'
> collect2: error: ld returned 1 exit status
[... cut ...]
>                                    ${CMAKE_CURRENT_SOURCE_DIR}/../lib
>if(BUILD_TESTS)
>				${CMAKE_CURRENT_SOURCE_DIR}/../app/tng
>endif()
>                                    ${CMAKE_CURRENT_SOURCE_DIR}/../third_party
>                                    ${CMAKE_CURRENT_SOURCE_DIR}/../third_party/mbedtls/include
>                                    ${CMAKE_CURRENT_SOURCE_DIR}/../third_party/wolfssl
>                                    ${CMAKE_CURRENT_BINARY_DIR}/../lib)

Thanks!

With my best regards,
Alex

bryan-hunt, 2023-08-02:
In your cmake command line add the following:
```
-DATCA_TNGTLS_SUPPORT=ON -DATCA_TNGLORA_SUPPORT=ON -DATCA_TFLEX_SUPPORT=ON -DATCA_TNG_LEGACY_SUPPORT=ON
```

fittila, 2023-10-02:
Everything works fine with suggested input parameters. Thanks!

closed 2023-10-02 by fittila