MicrochipTech/cryptoauthlib #343: Missing cryptoauthlib.conf/slot.conf.tmpl

ndivecha, 2023-08-03:
**Describe the bug**
This is my fourth compilation and on four different systems (x2 - Debian 11, PopOS, Ubuntu, Yocto Project). Following exact installation steps from [PKCS11-Linux-Setup](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) 

```
sudo apt-get install cmake libudev-dev
git clone https://github.com/MicrochipTech/cryptoauthlib
cd cryptoauthlib
cmake .
make
sudo make install 
```
It compiles OK, but fails to install cryptoauthlib.conf/slot.conf.tmpl. Got same issue while running Yocto layer with commit id 856a46b30aa03bca54f0e9402d1a30dccfe5aa8a. It fails to find slot.conf.tmpl.

Full output from freshly installed PopOS: 
```
nitesh@pop-os:~/Projects$ git clone https://github.com/MicrochipTech/cryptoauthlib
Cloning into 'cryptoauthlib'...
remote: Enumerating objects: 33344, done.
remote: Counting objects: 100% (4336/4336), done.
remote: Compressing objects: 100% (2073/2073), done.
remote: Total 33344 (delta 2375), reused 4044 (delta 2259), pack-reused 29008
Receiving objects: 100% (33344/33344), 70.04 MiB | 3.35 MiB/s, done.
Resolving deltas: 100% (26020/26020), done.
nitesh@pop-os:~/Projects$ cd cryptoauthlib
nitesh@pop-os:~/Projects/cryptoauthlib$ cmake .
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for __GNU_LIBRARY__
-- Looking for __GNU_LIBRARY__ - found
-- Looking for malloc
-- Looking for malloc - found
-- Looking for free
-- Looking for free - found
-- Looking for strcasestr
-- Looking for strcasestr - found
-- Looking for memset_s
-- Looking for memset_s - not found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nitesh/Projects/cryptoauthlib
nitesh@pop-os:~/Projects/cryptoauthlib$ make
[  1%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_basic.c.o
[  3%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_cfgs.c.o
[  5%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_debug.c.o
[  6%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_device.c.o
[  8%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_helpers.c.o
[ 10%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_iface.c.o
/home/nitesh/Projects/cryptoauthlib/lib/atca_iface.c: In function ‘ifacecfg_set_address’:
/home/nitesh/Projects/cryptoauthlib/lib/atca_iface.c:467:13: warning: unused parameter ‘addr’ [-Wunused-parameter]
  467 |     uint8_t addr,           /**< [in] Desired address */
      |     ~~~~~~~~^~~~
/home/nitesh/Projects/cryptoauthlib/lib/atca_iface.c:468:17: warning: unused parameter ‘kitiface’ [-Wunused-parameter]
  468 |     ATCAKitType kitiface    /**< [in] Optional parameter to set the kit iface type */
      |     ~~~~~~~~~~~~^~~~~~~~
[ 11%] Building C object lib/CMakeFiles/cryptoauth.dir/atca_utils_sizes.c.o
[ 13%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_aes.c.o
[ 15%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_aes_gcm.c.o
[ 16%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_basic.c.o
[ 18%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_checkmac.c.o
[ 20%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_command.c.o
[ 21%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_counter.c.o
[ 23%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_delete.c.o
[ 25%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_derivekey.c.o
[ 26%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_ecdh.c.o
[ 28%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_execution.c.o
[ 30%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_gendig.c.o
[ 31%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_genkey.c.o
[ 33%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_helpers.c.o
[ 35%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_hmac.c.o
[ 36%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_info.c.o
[ 38%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_kdf.c.o
[ 40%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_lock.c.o
[ 41%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_mac.c.o
[ 43%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_nonce.c.o
[ 45%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_privwrite.c.o
[ 46%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_random.c.o
[ 48%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_read.c.o
[ 50%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_secureboot.c.o
[ 51%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_selftest.c.o
[ 53%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_sha.c.o
[ 55%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_sign.c.o
[ 56%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_updateextra.c.o
[ 58%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_verify.c.o
[ 60%] Building C object lib/CMakeFiles/cryptoauth.dir/calib/calib_write.c.o
/home/nitesh/Projects/cryptoauthlib/lib/calib/calib_write.c: In function ‘calib_ca2_write_bytes_zone’:
/home/nitesh/Projects/cryptoauthlib/lib/calib/calib_write.c:872:34: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
  872 |         (block != 0) || ((length > 16 * (4-slot)) || ((length % 16) != 0))))
      |                                  ^
[ 61%] Building C object lib/CMakeFiles/cryptoauth.dir/host/atca_host.c.o
[ 63%] Building C object lib/CMakeFiles/cryptoauth.dir/__/app/api_206a/api_206a.c.o
[ 65%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_client.c.o
[ 66%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_date.c.o
[ 68%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_def.c.o
[ 70%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_der.c.o
[ 71%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_host_hw.c.o
[ 73%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_host_sw.c.o
[ 75%] Building C object lib/CMakeFiles/cryptoauth.dir/atcacert/atcacert_pem.c.o
[ 76%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_hw_aes_cbc.c.o
[ 78%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_hw_aes_cbcmac.c.o
[ 80%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_hw_aes_ccm.c.o
[ 81%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_hw_aes_cmac.c.o
[ 83%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_hw_aes_ctr.c.o
[ 85%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_pad.c.o
[ 86%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_pbkdf2.c.o
[ 88%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_sw_sha1.c.o
[ 90%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/atca_crypto_sw_sha2.c.o
[ 91%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/hashes/sha1_routines.c.o
[ 93%] Building C object lib/CMakeFiles/cryptoauth.dir/crypto/hashes/sha2_routines.c.o
[ 95%] Building C object lib/CMakeFiles/cryptoauth.dir/jwt/atca_jwt.c.o
[ 96%] Building C object lib/CMakeFiles/cryptoauth.dir/hal/atca_hal.c.o
[ 98%] Building C object lib/CMakeFiles/cryptoauth.dir/hal/hal_linux.c.o
[100%] Linking C shared library libcryptoauth.so
[100%] Built target cryptoauth
nitesh@pop-os:~/Projects/cryptoauthlib$ 

nitesh@pop-os:~$ ls /etc/cryptoauthlib/cryptoauthlib.conf
ls: cannot access '/etc/cryptoauthlib/cryptoauthlib.conf': No such file or directory
nitesh@pop-os:~$ ls /var/lib/cryptoauthlib/slot.conf.tmpl
ls: cannot access '/var/lib/cryptoauthlib/slot.conf.tmpl': No such file or directory
```

Any help will be highly appreciated. 

Cheers, 
Nitesh

bryan-hunt, 2023-08-03:
You didn't execute the last step listed in the [instructions](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup#build-and-install-cryptoauthlib-with-pkcs11-support): `$ sudo make install`

bryan-hunt, 2023-08-03:
PKCS11 isn't enabled by default. Enable it by adding `-DATCA_PKCS11=ON` to your cmake invocation.

I updated the wiki to reflect this.

ndivecha, 2023-08-03:
Worked like a charm on PopOS... I see now both files are installed. I will give a try with Yocto and other systems. 

Much appreciate for your support. 

```
nitesh@pop-os:~/Projects/cryptoauthlib$ sudo make install
Consolidate compiler generated dependencies of target cryptoauth
[100%] Built target cryptoauth
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Installing: /etc/cryptoauthlib
-- Installing: /etc/cryptoauthlib/cryptoauthlib.conf
-- Installing: /var/lib/cryptoauthlib
-- Installing: /var/lib/cryptoauthlib/slot.conf.tmpl
-- Installing: /usr/lib/libcryptoauth.so
-- Up-to-date: /usr/include/cryptoauthlib/atca_basic.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_bool.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_cfgs.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_compiler.h
-- Installing: /usr/include/cryptoauthlib/atca_config.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_config_check.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_debug.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_device.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_devtypes.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_helpers.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_iface.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_platform.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_status.h
-- Up-to-date: /usr/include/cryptoauthlib/atca_version.h
-- Up-to-date: /usr/include/cryptoauthlib/cryptoauthlib.h
-- Installing: /usr/include/cryptoauthlib/atca_config.h
-- Up-to-date: /usr/include/cryptoauthlib/hal/atca_hal.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_check_config.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_client.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_date.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_def.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_der.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_host_hw.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_host_sw.h
-- Up-to-date: /usr/include/cryptoauthlib/atcacert/atcacert_pem.h
-- Up-to-date: /usr/include/cryptoauthlib/calib/calib_aes_gcm.h
-- Up-to-date: /usr/include/cryptoauthlib/calib/calib_basic.h
-- Up-to-date: /usr/include/cryptoauthlib/calib/calib_command.h
-- Up-to-date: /usr/include/cryptoauthlib/calib/calib_config_check.h
-- Up-to-date: /usr/include/cryptoauthlib/calib/calib_execution.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/atca_crypto_hw_aes.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/atca_crypto_sw.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/atca_crypto_sw_sha1.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/atca_crypto_sw_sha2.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/crypto_config_check.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/hashes/sha1_routines.h
-- Up-to-date: /usr/include/cryptoauthlib/crypto/hashes/sha2_routines.h
-- Up-to-date: /usr/include/cryptoauthlib/host/atca_host.h
-- Up-to-date: /usr/include/cryptoauthlib/host/atca_host_config_check.h
-- Up-to-date: /usr/include/cryptoauthlib/jwt/atca_jwt.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/cryptoki.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_attrib.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_cert.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_debug.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_digest.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_encrypt.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_find.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_info.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_init.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_key.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_mech.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_object.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_os.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_session.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_signature.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_slot.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_token.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_util.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11f.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11t.h
-- Installing: /usr/include/cryptoauthlib/pkcs11/pkcs11_config.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tflxtls_cert_def_4_device.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tng_atca.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tng_atcacert_client.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tng_root_cert.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tnglora_cert_def_1_signer.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tnglora_cert_def_2_device.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tnglora_cert_def_4_device.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tngtls_cert_def_1_signer.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tngtls_cert_def_2_device.h
-- Up-to-date: /usr/include/cryptoauthlib/app/tng/tngtls_cert_def_3_device.h
-- Up-to-date: /usr/include/cryptoauthlib/app/wpc/atca_config.h
-- Up-to-date: /usr/include/cryptoauthlib/app/wpc/wpc_apis.h
-- Up-to-date: /usr/include/cryptoauthlib/app/wpc/wpc_check_config.h
-- Up-to-date: /usr/include/cryptoauthlib/app/wpc/wpccert_client.h
-- Up-to-date: /usr/include/cryptoauthlib/app/wpc/zcust_def_1_signer.h
-- Up-to-date: /usr/include/cryptoauthlib/app/wpc/zcust_def_2_device.h
-- Up-to-date: /usr/include/cryptoauthlib/app/api_206a/api_206a.h
nitesh@pop-os:~/Projects/cryptoauthlib$ 
```

closed 2023-08-03 by bryan-hunt

ndivecha, 2023-08-04:
@bryan-hunt 

Thanks for all your support so far... I don't know if you can help or guide with issue in Yocto. Having similar problem and I did enable following `cmake` flags in my .bbappend file also: 

```
EXTRA_OECMAKE += " \
    -DATCA_HAL_I2C=ON \
    -DATCA_PKCS11=ON \
    -DATCA_OPENSSL=ON \
    -DATCA_ATECC508A_SUPPORT=ON \
    -DATCA_ATECC608_SUPPORT=ON \
    -DATCA_BUILD_SHARED_LIBS=ON \
    -DATCA_TNGTLS_SUPPORT=ON \
    -DATCA_TNGLORA_SUPPORT=ON \
    -DATCA_TFLEX_SUPPORT=ON \
    -DATCA_USE_ATCAB_FUNCTIONS=ON \
"
```

It downloads new the newest commit id `86b3046d2e` but while installing it fails to find `slot.conf.tmpl`. 

```
ERROR: cryptoauthlib-1.0+gitAUTOINC+86b3046d2e-r0 do_install: ExecutionError('/home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/run.do_install.3384700', 1, None, None)
ERROR: Logfile of failure stored in: /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/temp/log.do_install.3384700
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-connectivity/openssl/openssl_3.0.8.bb:do_populate_sysroot', '/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-core/glibc/glibc_2.35.bb:do_populate_sysroot', '/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-core/systemd/systemd_250.5.bb:do_populate_sysroot', '/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/cmake/cmake-native_3.22.3.bb:do_populate_sysroot', '/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/gcc/gcc-cross_11.3.bb:do_populate_sysroot', '/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_11.3.bb:do_populate_sysroot', '/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/nitesh/Projects/meta-local-kirkstone/build/../openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['openssl', 'glibc', 'systemd', 'cmake-native', 'gcc-cross-arm', 'gcc-runtime', 'quilt-native', 'ninja-native', 'patch-native', 'pseudo-native', 'linux-libc-headers', 'opkg-utils', 'libtool-native', 'attr-native', 'xz-native', 'zstd-native', 'binutils-cross-arm', 'libmpc-native', 'zlib-native', 'mpfr-native', 'texinfo-dummy-native', 'flex-native', 'gnu-config-native', 'gmp-native', 'shadow', 'acl', 'libcap', 'libseccomp', 'bash-completion', 'util-linux', 'kmod', 'zstd', 'shadow-sysroot', 'base-files', 'base-passwd', 'shadow-native', 'curl-native', 'ncurses-native', 'bzip2-native', 'libgcc', 're2c-native', 'gettext-minimal-native', 'm4-native', 'attr', 'libxcrypt', 'zlib', 'libcap-ng', 'util-linux-libuuid', 'ncurses', 'xz', 'openssl-native', 'perl-native', 'make-native', 'gdbm-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| NOTE: DESTDIR=/home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image VERBOSE=1 cmake --build /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/build --target install --
| [0/1] cd /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/build && /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/recipe-sysroot-native/usr/bin/cmake -P cmake_install.cmake
| -- Install configuration: "RelWithDebInfo"
| -- Installing: /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/etc/cryptoauthlib
| -- Installing: /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/var/lib/cryptoauthlib
| -- Installing: /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/usr/lib/libcryptoauth.so
| -- Installing: /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/usr/include/cryptoauthlib/atca_basic.h
| -- Installing: /home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/usr/include/cryptoauthlib/atca_bool.h
[... cut ...]
| cp: cannot stat '/home/nitesh/Projects/meta-local-kirkstone/build/tmp/work/cortexa7hf-neon-vfpv4-eboot-linux-gnueabi/cryptoauthlib/1.0+gitAUTOINC+86b3046d2e-r0/image/var/lib/cryptoauthlib/slot.conf.tmpl': No such file or directory
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/nitesh/Projects/meta-local-kirkstone/build/../meta-atmel/recipes-security/cryptoauthlib/cryptoauthlib_git.bb:do_install) failed with exit code '1'
```

Any help/insights will be highly appreciated. 

Cheers, 
Nitesh

bryan-hunt, 2023-08-04:
For yocto you wouldn't do an install step. I would suggest contacting the recipe maintainer.

ndivecha, 2023-08-04:
@bryan-hunt 

Roger... much appreciated.