MicrochipTech/cryptoauthlib #344: pkcs11 branch is missing

0xaaa111, 2023-08-14:
**Describe the bug**

According to the [cryptoauthlib-manual.pdf](https://github.com/MicrochipTech/cryptoauthlib/blob/main/cryptoauthlib-manual.pdf) , PKCS support is implemented on branch **pkcs11**. But this branch is not presented in repository.

**To Reproduce**
1.See cryptoauthlib-manual.pdf page 12
2.git clone --single-branch -b pkcs11 https://github.com/MicrochipTech/cryptoauthlib
3.cloning fails "fatal: Remote branch pkcs11 not found in upstream origin"

**Expected behavior**
[image]

bryan-hunt, 2023-08-14:
PKCS11 is a mainline feature that is simply enabled with the cmake option `ATCA_PKCS11`. Please see the wiki: https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11

We'll update the doc generation.

closed 2023-08-14 by bryan-hunt

0xaaa111, 2023-08-14:
@bryan-hunt Thank you