MicrochipTech/cryptoauthlib #352: ATECC608B fail to test with cryptoauth_test

Psychesnet, 2023-09-25:
Hi Microchip,

We are using ATECC608B to achieve security boot feature at ARM base camera. We try to use cryptoauth_test to do testing first.
Following are the compile options we used. We base on i2c to do testing.

```
  CMAKE_OPTIONS+=-DBUILD_TESTS=ON \
                 -DATCA_PKCS11=ON \
                 -DATCA_OPENSSL=ON \
                 -DATCA_HAL_I2C=ON \
                 -DATCA_TNGTLS_SUPPORT=ON \
                 -DATCA_TFLEX_SUPPORT=ON \
                 -DATCA_TNG_LEGACY_SUPPORT=ON \
                 -DATCA_PRINTF=ON \
                 -DSETUP_INSTALLER=ON
```
               
We use i2cdect and we can get device is online.

```
root@agave-936b:/etc/wifi# i2cdetect -r -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- 22 -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --
```

We try to run some tests, it is failure.

```
root@agave-936b:/etc/wifi# cryptoauth_test info -d ecc608 -i i2c 1 -a 0xc0
 
./calib/calib_read.c:98:f0:calib_read_zone - execution failed
./calib/calib_read.c:333:f0:calib_read_zone - falied
atcab_init() failed with ret=0x000000F0
 
root@agave-936b:/etc/wifi# cryptoauth_test sernum -d ecc608 -i i2c 1 -a 0xc0
 
./calib/calib_read.c:98:f0:calib_read_zone - execution failed
./calib/calib_read.c:333:f0:calib_read_zone - falied
atcab_init() failed with ret=0x000000F0
```

In the other hand, we implement very simple code to do sernum function and link Cryptoauthlib, just like cryptoauth_test.

```
ATCAIfaceCfg cfg_ateccx08a_i2c = {
      .iface_type                 = ATCA_I2C_IFACE,
      .devtype                    = ATECC608A,
      {
          .atcai2c.address  = 0xC0,
          .atcai2c.bus            = 1,
          .atcai2c.baud = 100000,
      },
      .wake_delay                 = 1500,
      .rx_retries                 = 20
  };
  int main(int argc, char *argv[])
  {
      atcab_init(&cfg_ateccx08a_i2c);
      int ret = -1;
      uint8_t serialnum[ATCA_SERIAL_NUM_SIZE];
      char displaystr[30];
      size_t displaylen = sizeof(displaystr);
      do {
          if (atcab_read_serial_number(serialnum) != ATCA_SUCCESS) {                             
              fprintf(stderr, "fail to atcab sernum\n");
              break;                 
          } else {                                            
              atcab_bin2hex(serialnum, ATCA_SERIAL_NUM_SIZE, displaystr, &displaylen);
              printf("serial number:\r\n%s\r\n", displaystr);
          }                                                                   
          ret = 0;                                                                         
      } while (false);  
      atcab_release();
      return ret;
  }
```

This is the output, looks like anything is good.

```
root@AAA:/# ./catool
serial number:
01 23 3C 8E 93 F7 31 69 EE
```

My question is why cryptoauth_test fail to run? but my example code is okay? Please help. Thanks.

InvisibleGit, 2023-09-25:
See https://github.com/MicrochipTech/cryptoauthlib/issues/350 ? :)

closed 2023-09-25 by Psychesnet

Psychesnet, 2023-09-25:
Hi @InvisibleGit 

May I communicate with you via gmail? I have some questions about ECC608 to ask. Thanks.

InvisibleGit, 2023-09-25:
I mean, you can try... but:

- I'm don't have too much related knowledge. I'm a user of it just like you and I'm not related to Microchip in any way
- I've only used the TNG variant of the device which is probably the most common in embedded IoT projects, but I have no clue about FLEX and custom variants regarding the configuration zone locking and certificate slots (TNG is pre-provisioned at manufacturing).

So my recommendation would be to create an new issue and reap the benefits of other people who used the device and this would also leave a knowledge breadcrumb for future readers of similar problem(s).

I might chime in if I know some answer.

InvisibleGit, 2023-09-25:
Ahm... yes it's an easy question but the answer is quite elaborate. You need to learn how those devices work internally and specifically how to "finalize" or lock zones on your device, which is probably a FLEX variant since data zone is not yet locked. Locking should be done after you understand what it means (you could easily make your device unusable).

Few pointers:
- In the end you'll use atcab_lock(). See: https://microchiptech.github.io/cryptoauthlib/a19708.html#ga25d2895192d93c524f5dba92a492cdfb *but*
- Read Microchip docs first, they are of good quality and explain a lot
- https://www.instructables.com/Secure-Communication-Arduino/ (Ignore Arduino and that it mentiones A variant of ECC608, it doesn't matter)
- Another try is https://forum.digikey.com/t/atecc608-node-authentication-example-walk-through/13141/1 (again A variant is the same as B, don't let it confuse you)

Psychesnet, 2023-09-25:
Hi @InvisibleGit 

Thanks. I will study it now.