MicrochipTech/cryptoauthlib #353: Compilation fails when ATCA_USE_SHARED_MUTEX is off

jens-totemic, 2023-09-25:
When `ATCA_USE_SHARED_MUTEX` is not set in CMakelist, the compilation fails with 

```
lib/hal/hal_linux.c:292:13: error: conflicting types for 'hal_create_mutex'
ATCA_STATUS hal_create_mutex(void ** ppMutex, char* pName)
            ^
lib/hal/atca_hal.h:245:13: note: previous declaration is here
ATCA_STATUS hal_create_mutex(void ** ppMutex, const char* pName);
            ^
```

This is caused by https://github.com/MicrochipTech/cryptoauthlib/blob/main/lib/hal/hal_linux.c#L292
missing the `const` definition:
```
ATCA_STATUS hal_create_mutex(void ** ppMutex, char* pName)
```
needs to be:
```
ATCA_STATUS hal_create_mutex(void ** ppMutex, const char* pName)
```

jens-totemic, 2023-11-25:
Issues still exists in master

jens-totemic, 2024-01-26:
The issue still exists. Can this please be fixed?

KalyanCManukonda, 2024-03-24:
Hi, this had been fixed in the recent versions.

closed 2024-03-24 by KalyanCManukonda