MicrochipTech/cryptoauthlib #369: symmetric_authentication.c calculation mismatch (derived key vs. master key)

electronicayciencia, 2024-03-24:
Hello.

If I'm not wrong, in the symmetric authentication example app you calculate the MAC from a derived symmetric key, when the device is using the master key:

- you issue a Nonce command, then use the random output to locally calculate TempKey.
- you locally calculate the derived symmetric key
- you issue a MAC command, the device calculates the MAC using the master key slot and master key data,
- you locally compute the MAC, but using the derived key, so they never match.

In my tests, if you change line 108:

https://github.com/MicrochipTech/cryptoauthlib/blob/405d8311a203e61657597c81b23ae7a370f485d9/app/ip_protection/symmetric_authentication.c#L105-L112

to this:

```c
mac_params.key = master_key;
```

it works fine.

What did I miss?

KalyanCManukonda, 2024-03-24:
Thank you for reporting the observation. Please submit a request through the [myMicrochip Support portal](https://microchipsupport.force.com/s/). For details on the myMicrochip Support Portal or how to use it please see the [instructions for using the myMicrochip Support portal](https://microchipsupport.force.com/s/article/How-to-submit-a-case).

electronicayciencia, 2024-03-25:
Doesn't matter, thanks for the quick response. I'm just reporting it, in case someone encounters the same problem in the future.

closed 2024-03-25 by electronicayciencia

AriyaKrishnanAK, 2024-08-12:
Hi @electronicayciencia, in the symmetric authentication example app you need the master key at the host do derive the symmetric key. The symmetric key(derived from master key) is used for authentication. The example app is correct. I feel  you are missing the Master key that is available during provisioning of the device.  Please refer https://developerhelp.microchip.com/xwiki/bin/view/applications/security/symmetric-use-case-example/

electronicayciencia, 2024-09-30:
You're right. Thank you! To make it work I just had to provision the device with a previously calculated Derived key instead of the Master key.