MicrochipTech/cryptoauthlib #379: cryptoauth target does not propagate header path to other projects

jacobschloss, 2024-06-19:
**Describe the bug**
The cmake target "cryptoauth" does not declare its headers with target_include_directories, so other projects that use this library do not automatically pick up all of the needed header paths when linking wiht target_include_directories.

**To Reproduce**
Create a new cmake project that uses this library as a submodule. Create a small executable and link with target_link_libraries(foo PUBLIC cryptoauth).

As the needed header dirs are not exported with the cryptoauth target, the build will fail with 
fatal error: cryptoauthlib.h: No such file or directory

**Expected behavior**
Clean build

**Additional context**
Changing include_directories to target_include_directories at cryptoauthlib/lib/CMakeLists.txt
  https://github.com/MicrochipTech/cryptoauthlib/blob/405d8311a203e61657597c81b23ae7a370f485d9/lib/CMakeLists.txt#L309

seems to work, the headers are correctly propagated to another exe that links to cryptoauth and it builds without manually setting the header search path. May also need to change something for OPENSSL support / L326 if ATCA_OPENSSL is set.

jacobschloss, 2024-06-19:
The current call to include_directories seems to add a non-existent "cryptoauth" dir to the path for this file scope. I assume this isn't needed.

Srinivas-E, 2024-06-25:
Hi @jacobschloss 
Thanks for sharing your observations.
We shall check this and review for the required changes.
Since the downstream sources needs to be in sync, we shall absorb it as per planned our planned release process

jacobschloss, 2024-08-24:
> This issue has been marked as stale - please confirm the issue still exists with the latest version of the library and update the issue if it remains

yes

AriyaKrishnanAK, 2024-09-09:
Hi @jacobschloss,

You can use the make install command to install the cryptoauth library. Here are the steps to do so and how to link the library with your project:

Step 1: Install the cryptoauth Library
Navigate to the cryptoauth Library Directory: Open a terminal and navigate to the directory where the cryptoauth library is located.
Copy sh

   cd path/to/cryptoauth
Build and Install the Library: Run the following commands to build and install the cryptoauth library. This will copy the necessary files to your system's library and include directories.
Copy sh

   mkdir build
   cd build
   cmake ..
   make
   sudo make install
The sudo make install command may require administrative privileges, depending on your system configuration.

Step 2: Link the cryptoauth Library with Your Project
Create a New CMake Project: Create a new CMake project or navigate to your existing project directory.

Modify Your CMakeLists.txt File: Update your CMakeLists.txt file to find and link the cryptoauth library.

jacobschloss, 2024-09-11:
Yes, but this way I do not need to install the library and the list of directories needed is picked up by CMake in my parent project. I am building cryptoauthlib from source as a cmake sub-project as we are building it for a matrix of 3 OS targets and 3 cpu architectures (eg, 9 build targets, 27 build targets including debug and MinSizeRel) and do I not want to manage archiving and distributing binaries separately for this library. It is much easier to handle as an in-tree source dependency as it compiles fast.

I have a local fork of cryptoauthlib with a patch to use target_include_directories applied. My feedback is just that is is normal for cmake library build targets to set their include directories with target_include_directories so that the list of required headers is propagated upwards to consuming projects.

jacobschloss, 2024-09-11:
What I mean is using target_include_directories instead of include_directories will make building as a nested cmake project work, which is a common and useful set up especially when dealing with multi-arch.

closed 2024-11-17 by github-actions[bot]