MicrochipTech/cryptoauthlib #396: ATECC608B-TNGTLS

Jeroen-Lardot-ACA, 2024-11-21:
Hey everyone,

I'm trying to get the ATECC608B working on greengrass through cryptoauthlib, but am running into some issues with authentication when running greengrass that I can't seem to get around. I tried changing pins etc, but that doesn't seem possible on the TNG variant. Does anyone have any idea on the underlying issue? AWS certificates and policies have been attached to the thing through the manifest file as provided by Microchip.

This is the module setup:

```
root@blox-rema:\~# p11tool --provider=/usr/lib/libcryptoauth.so --list-tokens
Token 0:
URL: pkcs11:model=ATECC608B;manufacturer=Microchip%20Technology%20Inc;serial=ABCDEFGH;token=MCHP
Label: MCHP
Type: Hardware token
Flags: RNG, uPIN uninitialized
Manufacturer: Microchip Technology Inc
Model: ATECC608B
Serial: ABCDEFGH
Module:

root@blox-rema:\~# p11tool --provider=/usr/lib/libcryptoauth.so --list-all
Object 0:
URL: pkcs11:model=ATECC608B;manufacturer=Microchip%20Technology%20Inc;serial=ABCDEFGH;token=MCHP;id=ABCDEFGH;object=device;type=private
Type: Private key (EC/ECDSA-SECP256R1)
Label: device
Flags: CKA_PRIVATE; CKA_NEVER_EXTRACTABLE; CKA_SENSITIVE;
ID: ABCDEFGH

Object 1:
URL: pkcs11:model=ATECC608B;manufacturer=Microchip%20Technology%20Inc;serial=ABCDEFGH;token=MCHP;id=ABCDEFGH;object=device;type=public
Type: Public key (EC/ECDSA-SECP256R1)
Label: device
ID: ABCDEFGH
```

The greengrass config (without nucleus):
```
system:
  certificateFilePath: "pkcs11:object=device;type=cert"
  privateKeyPath: "pkcs11:object=device;type=private"
  rootCaPath: "/greengrass/v2/config/AmazonRootCA1.pem"
  rootpath: "/greengrass/v2"
  thingName: "0123B87E3C22BD5D01"
services:
  aws.greengrass.crypto.Pkcs11Provider:
    configuration:
      library: "/usr/lib/libcryptoauth.so"
      name: "lybcryptauth_pkcs11"
      slot: 0
      userPin: 1234
    dependencies: []
    version: "0.0.0"
```

The error I'm getting:
```
24-11-21T16:33:44.921Z [INFO] (pool-3-thread-14) com.aws.greengrass.security.SecurityService: Register crypto key service provider. {keyType=pkcs11}
2024-11-21T16:33:44.921Z [INFO] (pool-3-thread-14) com.aws.greengrass.security.SecurityService: Register MQTT connection security provider. {keyType=pkcs11}
2024-11-21T16:33:44.922Z [INFO] (aws.greengrass.crypto.Pkcs11Provider-lifecycle) com.aws.greengrass.security.provider.pkcs11.PKCS11CryptoKeyService: service-set-state. {serviceName=aws.greengrass.crypto.Pkcs11Provider, currentState=STARTING, newState=RUNNING}
2024-11-21T16:33:44.923Z [INFO] (main-lifecycle) com.aws.greengrass.lifecyclemanager.GenericExternalService: service-set-state. {serviceName=main, currentState=INSTALLED, newState=STARTING}
2024-11-21T16:33:44.924Z [INFO] (main-lifecycle) com.aws.greengrass.lifecyclemanager.GenericExternalService: service-set-state. {serviceName=main, currentState=STARTING, newState=FINISHED}
2024-11-21T16:33:44.924Z [INFO] (pool-3-thread-8) com.aws.greengrass.lifecyclemanager.GenericExternalService: generic-service-finished. Nothing done. {serviceName=main, currentState=STARTING}
2024-11-21T16:33:44.927Z [INFO] (main-lifecycle) com.aws.greengrass.lifecyclemanager.GenericExternalService: service-set-state. {serviceName=main, currentState=FINISHED, newState=STOPPING}
2024-11-21T16:33:44.927Z [INFO] (pool-3-thread-8) com.aws.greengrass.lifecyclemanager.GenericExternalService: Shutdown initiated. {serviceName=main, currentState=STOPPING}
2024-11-21T16:33:44.927Z [INFO] (pool-3-thread-8) com.aws.greengrass.lifecyclemanager.GenericExternalService: generic-service-shutdown. {serviceName=main, currentState=STOPPING}
2024-11-21T16:33:44.928Z [INFO] (main-lifecycle) com.aws.greengrass.lifecyclemanager.GenericExternalService: service-set-state. {serviceName=main, currentState=STOPPING, newState=FINISHED}
2024-11-21T16:33:45.533Z [ERROR] (pool-3-thread-6) com.aws.greengrass.mqttclient.MqttClient: Error subscribing. {topic=$aws/things/0123B87E3C22BD5D01/jobs/$next/namespace-aws-gg-deployment/get/accepted}
java.util.concurrent.CompletionException: software.amazon.awssdk.crt.mqtt.MqttException: Error during getting mqtt connection builder
	at java.base/java.util.concurrent.CompletableFuture.encodeThrowable(CompletableFuture.java:331)
```

Thanks a lot for anyone providing any insights!

majh, 2024-12-10:
[this issue](https://github.com/MicrochipTech/cryptoauthlib/issues/328#issuecomment-1418661139) may be relevant.

closed 2025-02-16 by github-actions[bot]

leonau-amzn, 2025-09-09:
Thank you for reopening the issue.

Let me know if there is any information that I can provide that can be of use.  I'm doing some prototyping with the ATECC608B-TNGTLS chip and trying to run Greengrass v2 on top and getting the same set of failures as above (was torn on whether to open a new issue, but I took a chance at reusing the same issue so that there is one thing to follow).

leonau-amzn, 2026-01-09:
Please unmark this issue from being Stale.  This is still unresolved.

Do we have an ETA for the upcoming release?

leonau-amzn, 2026-02-05:
Sure, let me see what I can add.  I'm open to going to the very tip of the main branch / latest release if that's where the fix will be, since I'd like to stay on the latest and greatest anyway.

To summarize the background:

1. @JamieHunter added support Java and Greengrass v2 to cryptoauthlib after the release of v3.3 (right before v3.4) on this commit (https://github.com/MicrochipTech/cryptoauthlib/commit/9a37b8d685c351faed720d18d9610345bdaf20b0). I was able to get Greengrass v2 on Java working on this commit.
2. This feature worked on the next 2 commits, and then the 3rd commit afterwards was a huge v3.4 release drop (https://github.com/MicrochipTech/cryptoauthlib/commit/e723fd0a) - this commit is where I no longer was able to get Greengrass to work.

On my setup / reproducton:

Hardware:
1. Raspberry Pi 5 (with Raspberry Pi OS, the recommended default version from Raspberry Pi Imager. This is a pretty generic Debian 

Linux distribution)
2. ATECC608C chip (Trust-and-Go model)
3. UDFN socket board to connect the Raspberry Pi with the ATECC chip

Software:

To start with, I developed this using a Raspberry Pi 5 and the Raspberry Pi OS (recommended default version from Raspberry Pi Imager).  I'm using the UDFN socket board with attachment to the ATECC608C Trust-and-Go chip for access to the security chip. I've attached a shell script that I cobbled together that shows my setup - basically just following the various instructions on Microchip websites but with slightly updated dependencies because some docs started getting outdated (it's been a few months, so I don't remember exactly which parts).

[setup_atecc608.sh](https://github.com/user-attachments/files/25103568/setup_atecc608.sh)

After verifying the ATECC chip setup, I followed the manual installation instructions on https://docs.aws.amazon.com/greengrass/v2/developerguide/manual-installation.html.  Note that the recommendation on the website to use the `/greengrass/v2` folder is a requirement, not optional, as far as I could tell in my experiments.

Srinivas-E, 2026-02-18:
Hi @leonau-amzn ,
Thanks again for sharing your setup details. Sorry that I could not be able to spend much time to root cause this issue.
However, I tried a few simple steps to check tls connect establishments on CALv3.3, CALv3.4 an then using latest v3.8.0.
I notice and concur with your observations about v3.3 versus v3.4. But, I notice that when switched to latest v3.8.0, could get a proper TLS connection establishment!

[cal_tls_connect.log](https://github.com/user-attachments/files/25397906/cal_tls_connect.log) shows some more details on this.

In addition, I tried to spend some time on Greengrass/v2 setup, though could not get it right where I suspect connection handshakes were past through TLS, 

[socket_close_issue.log](https://github.com/user-attachments/files/25398004/socket_close_issue.log). 

Hence I resorted to openssl based tls checks to record my observations.
Could you take a look if this feedback helps to get past your issue?

Attached [pub_pkcs11.py](https://github.com/user-attachments/files/25398071/pub_pkcs11.py) for reference.

leonau-amzn, 2026-02-20:
Thank you for the update. Let me pull the latest from v3.8.0 to test it out and see if things somehow got fixed in the middle because of another change.  I don't have access to my device right now, will target to give it a quick test by 2/24.

Srinivas-E, 2026-02-24:
In addition, could you please create a ticket at https://www.microchip.com/en-us/support so that a relevant team member may be able to help you with a more directed conversation/support on this, including file/log share easily.