MicrochipTech/cryptoauthlib #398: Why does the sha2_routines.c: sw_sha256_process function use ATCA_UINT32_HOST_TO_LE by default?

xingyi-engine, 2025-03-07:
**Describe the bug**
A clear and concise description of what the bug is.

When I ported cryptolib to stm32, everytime I try to use atcab_ecdh_ioenc, the generated premaster key was wrong. Then I found sha2_routines.c: sw_sha256_process function which used ATCA_UINT32_HOST_TO_LE by default. When I change it to ATCA_UINT32_HOST_TO_BE,  the result is right. But in the SHA-256 standard defined in RFC 4634, the default is big-endian byte order to interpret data。

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

Srinivas-E, 2025-03-11:
Hi @xingyi-engine 
As sw_crypto is executed in the host environment/MCU where the library resides, our tests on a Windows host platform could not reproduce such issue. We shall look into this and take it for further actions.
Thank you for bringing this note and we shall keep you posted.

AriyaKrishnanAK, 2025-04-28:
Hi @xingyi-engine,

Thank you for your detailed feedback regarding the behavior you observed when porting CryptoLib to the STM32 platform.

Upon reproducing the environment using QEMU for a big-endian system and porting the library accordingly, we were able to confirm that the SHA-2 implementation (sw_sha256_process) behaves correctly according to the specification when properly handling byte order.

In addition, for SHA processing, the message block is explicitly byte-swapped during loading. This ensures that the word values (w_union.w_dword) are always in big-endian format, regardless of the system's native endianness (little or big endian).
Thus, the SW implementation properly handles both little-endian and big-endian architectures consistently.

Please let me know in case you need any further assistance.

Regards,
Ariya

Srinivas-E, 2025-06-04:
> Hi [@xingyi-engine](https://github.com/xingyi-engine),
>
> Thank you for your detailed feedback regarding the behavior you observed when porting CryptoLib to the STM32 platform.
>
> Upon reproducing the [...]

Hi @xingyi-engine ,
Just checking if you have any feedback on our latest update, before resolving this task.
Pld do re-open or log a new one if you have any open items.

closed 2025-08-11 by github-actions[bot]